ptvoinfo/zigbee-configurable-firmware #238: Newer version has lower resolution results for pressure sensors BM280

aleksander000, 2023-10-04:
ver. 1.8 :
"linkquality":235,"pressure_l5":99460,
ver. 1.9 :
"linkquality":231,"pressure_l5":994,

in ver 2.9 resolution is worst, two digits less

glcos, 2023-10-04:
@aleksander000 I observed the same behaviour https://github.com/ptvoinfo/zigbee-configurable-firmware/issues/225#issuecomment-1697033227
Hope this will be fixed to have the same precision as the previous version

ptvoinfo, 2023-10-05:
The new version uses the standard Zigbee cluster for pressure and sends kPa to a coordinator with one digit after the decimal point. Z2M or other higher system may round this value. You can enable debug logs and check incoming raw data.

aleksander000, 2023-10-09:
Why the previous versions did not need any play with debug etc. and showed more precision data ?

ptvoinfo, 2023-10-19:
@aleksander000 Could you please try to enable debug log in Z2M and check data in incoming Zigbee message?

DMUTPUUM, 2024-02-14:
[image]

closed 2024-03-31 by ptvoinfo