ptvoinfo/zigbee-configurable-firmware #243: [Vote] Add support UART PM2.5 air quality sensor PM1006

MrC0nst, 2023-10-22:
Hi! please add support UART PM2.5 air quality sensor PM1006
This sensor is used in cheap IKEA VINDRIKTNING air quality sensor

Additional info:
https://www.fabiosoft.com/2021/12/02/ikea-hack-making-the-air-quality-sensor-vindriktning-smart-sensor/
https://esphome.io/components/sensor/pm1006.html
https://github.com/Hypfer/esp8266-vindriktning-particle-sensor
https://github.com/bertrik/pm1006

[PM1006_LED_PARTICLE_SENSOR_MODULE_SPECIFICATIONS-1.pdf](https://github.com/ptvoinfo/zigbee-configurable-firmware/files/13064185/PM1006_LED_PARTICLE_SENSOR_MODULE_SPECIFICATIONS-1.pdf)

dido8884, 2023-10-22:
Да неплохо добавить и psm5003, дороже конечно но и качество будет получше.

ptvoinfo, 2023-11-02:
Added in the latest version.

closed 2023-11-02 by ptvoinfo