ptvoinfo/zigbee-configurable-firmware #245: CC2652P in PSM cannot set proper role despite GUI config

glcos, 2023-10-28:
I have a battery operated device based on CC2652P that I use as a remote garage door sensor.
This is the PSM configuration with latest 1.9.0.8 firmware.

[image]

**Issue**: I always get the same cluster configuration in the receiving device (Home Assistant) no matter the configuraton in the GUI interface.
I attempted to change the ROLE setting as Occupancy, Generic, Contact, but nothing changes, I always get the same exact ZigBee signature.
When the device detects door open/closed I see that corresponding switch changes status, but this is not the expected behaviour.
Am I doing something wrong?

[image]

```
{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress: 128>, manufacturer_code=64447, maximum_buffer_size=80, maximum_incoming_transfer_size=160, server_mask=11264, maximum_outgoing_transfer_size=160, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=False, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "1": {
      "profile_id": "0x0104",
      "device_type": "0xfffe",
      "input_clusters": [
        "0x0000",
        "0x0001",
        "0x0007"
      ],
      "output_clusters": [
        "0x0000",
        "0x0001",
        "0x0012"
      ]
    },
    "2": {
      "profile_id": "0x0104",
      "device_type": "0xfffe",
      "input_clusters": [
        "0x0006",
        "0x0007"
      ],
      "output_clusters": [
        "0x0006"
      ]
    },
    "3": {
      "profile_id": "0x0104",
      "device_type": "0xfffe",
      "input_clusters": [
        "0x0006",
        "0x0007"
      ],
      "output_clusters": [
        "0x0006"
      ]
    },
    "4": {
      "profile_id": "0x0104",
      "device_type": "0xfffe",
      "input_clusters": [
        "0x0402"
      ],
      "output_clusters": []
    }
  },
  "manufacturer": "CSRLabs",
  "model": "remoteswitch",
  "class": "zigpy.device.Device"
}
```

ptvoinfo, 2023-10-28:
These options work only with Zigbee2MQTT and created a custom converter.

glcos, 2023-10-28:
OK, understand, I will continue to use a custom quirk on Home Assistant, you can close this.

closed 2023-10-30 by ptvoinfo