ptvoinfo/zigbee-configurable-firmware #248: only one ds18b20 sensor?

MarkStephan89, 2023-11-09:
Hello everyone,
I use espeasy on an esp32 to read 10 ds18b20 sensors. 

[image]

Since the WiFi is very bad, I wanted to use Zigbee.
(Zigbee network already exists)

I use this Zigbee device: (CC2531)

[image]

I get the 5V power and ground for the sensors from the USB pin.

and the following configuration:
```
[Common]
Left=-1230
Top=317
Width=1129
Board=5
Device type=1
LED active=True
LED inverted=False
Pin LED=P00
LED mode=2
Pin active 1=True
Pin No 1=P02
Pin Type 1=7
Pull Type 1=0
Pin inversed 1=False
StatechkState_1=True
Pin active 2=True
Pin No 2=P02
Pin Type 2=7
Pull Type 2=0
Pin inversed 2=False
StatechkState_2=True
Pin active 3=True
Pin No 3=P02
Pin Type 3=7
Pull Type 3=0
Pin inversed 3=False
StatechkState_3=True
Pin active 4=True
Pin No 4=P02
Pin Type 4=7
Pull Type 4=0
Pin inversed 4=False
StatechkState_4=True
Pin active 5=True
Pin No 5=P02
Pin Type 5=7
Pull Type 5=0
Pin inversed 5=False
StatechkState_5=True
Pin active 6=True
Pin No 6=P02
Pin Type 6=7
Pull Type 6=0
Pin inversed 6=False
StatechkState_6=True
Pin active 7=True
Pin No 7=P02
Pin Type 7=7
Pull Type 7=0
Pin inversed 7=False
StatechkState_7=True
Pin active 8=True
Pin No 8=P02
Pin Type 8=7
Pull Type 8=0
Pin inversed 8=False
StatechkState_8=True
Pin active 9=False
Pin No 9=
Pin Type 9=0
Pull Type 9=0
Pin inversed 9=False
StatechkState_9=True
Pin active 10=False
Pin No 10=
Pin Type 10=0
Pull Type 10=0
Pin inversed 10=False
StatechkState_10=True
Pin active 11=False
Pin No 11=
Pin Type 11=0
Pull Type 11=0
Pin inversed 11=False
StatechkState_11=True
Pin active 12=False
Pin No 12=
Pin Type 12=0
Pull Type 12=0
Pin inversed 12=False
StatechkState_12=True
Pin active 13=False
Pin No 13=
Pin Type 13=0
Pull Type 13=0
Pin inversed 13=False
StatechkState_13=True
Pin active 14=False
Pin No 14=
Pin Type 14=0
Pull Type 14=0
Pin inversed 14=False
StatechkState_14=True
Pin active 15=False
Pin No 15=
Pin Type 15=0
Pull Type 15=0
Pin inversed 15=False
StatechkState_15=True
Pin active 16=False
Pin No 16=
Pin Type 16=0
Pull Type 16=0
Pin inversed 16=False
StatechkState_16=True
Pin active 17=False
Pin No 17=
Pin Type 17=0
Pull Type 17=0
Pin inversed 17=False
LinkchkLink_17=True
SwitchchkSwitch_17=False
HoldchkHold_17=False
DoublechkDouble_17=False
TripplechkTripple_17=False
Bind type 17=0
Pin active 18=False
Pin No 18=
Pin Type 18=0
Pull Type 18=0
Pin inversed 18=False
LinkchkLink_18=True
SwitchchkSwitch_18=False
HoldchkHold_18=False
DoublechkDouble_18=False
TripplechkTripple_18=False
Bind type 18=0
Pin active 19=False
Pin No 19=
Pin Type 19=0
Pull Type 19=0
Pin inversed 19=False
LinkchkLink_19=True
SwitchchkSwitch_19=False
HoldchkHold_19=False
DoublechkDouble_19=False
TripplechkTripple_19=False
Bind type 19=0
Pin active 20=False
Pin No 20=
Pin Type 20=0
Pull Type 20=0
Pin inversed 20=False
LinkchkLink_20=True
SwitchchkSwitch_20=False
HoldchkHold_20=False
DoublechkDouble_20=False
TripplechkTripple_20=False
Bind type 20=0
Pin active 21=False
Pin No 21=
Pin Type 21=0
Pull Type 21=0
Pin inversed 21=False
LinkchkLink_21=True
SwitchchkSwitch_21=False
HoldchkHold_21=False
DoublechkDouble_21=False
TripplechkTripple_21=False
Bind type 21=0
Pin active 22=False
Pin No 22=
Pin Type 22=0
Pull Type 22=0
Pin inversed 22=False
LinkchkLink_22=True
SwitchchkSwitch_22=False
HoldchkHold_22=False
DoublechkDouble_22=False
TripplechkTripple_22=False
Bind type 22=0
Pin active 23=False
Pin No 23=
Pin Type 23=0
Pull Type 23=0
Pin inversed 23=False
LinkchkLink_23=True
SwitchchkSwitch_23=False
HoldchkHold_23=False
DoublechkDouble_23=False
TripplechkTripple_23=False
Bind type 23=0
Pin active 24=False
Pin No 24=
Pin Type 24=0
Pull Type 24=0
Pin inversed 24=False
LinkchkLink_24=True
SwitchchkSwitch_24=False
HoldchkHold_24=False
DoublechkDouble_24=False
TripplechkTripple_24=False
Bind type 24=0
Pin active 25=False
Pin No 25=
Pin Type 25=0
Pull Type 25=0
Pin inversed 25=False
LinkchkLink_25=True
SwitchchkSwitch_25=False
HoldchkHold_25=False
DoublechkDouble_25=False
TripplechkTripple_25=False
Bind type 25=0
Pin active 26=False
Pin No 26=
Pin Type 26=0
Pull Type 26=0
Pin inversed 26=False
LinkchkLink_26=True
SwitchchkSwitch_26=False
HoldchkHold_26=False
DoublechkDouble_26=False
TripplechkTripple_26=False
Bind type 26=0
Pin active 27=False
Pin No 27=
Pin Type 27=0
Pull Type 27=0
Pin inversed 27=False
LinkchkLink_27=True
SwitchchkSwitch_27=False
HoldchkHold_27=False
DoublechkDouble_27=False
TripplechkTripple_27=False
Bind type 27=0
Pin active 28=False
Pin No 28=
Pin Type 28=0
Pull Type 28=0
Pin inversed 28=False
LinkchkLink_28=True
SwitchchkSwitch_28=False
HoldchkHold_28=False
DoublechkDouble_28=False
TripplechkTripple_28=False
Bind type 28=0
Pin active 29=False
Pin No 29=
Pin Type 29=0
Pull Type 29=0
Pin inversed 29=False
LinkchkLink_29=True
SwitchchkSwitch_29=False
HoldchkHold_29=False
DoublechkDouble_29=False
TripplechkTripple_29=False
Bind type 29=0
Pin active 30=False
Pin No 30=
Pin Type 30=0
Pull Type 30=0
Pin inversed 30=False
LinkchkLink_30=True
SwitchchkSwitch_30=False
HoldchkHold_30=False
DoublechkDouble_30=False
TripplechkTripple_30=False
Bind type 30=0
Pin active 31=False
Pin No 31=
Pin Type 31=0
Pull Type 31=0
Pin inversed 31=False
LinkchkLink_31=True
SwitchchkSwitch_31=False
HoldchkHold_31=False
DoublechkDouble_31=False
TripplechkTripple_31=False
Bind type 31=0
Pin active 32=False
Pin No 32=
Pin Type 32=0
Pull Type 32=0
Pin inversed 32=False
LinkchkLink_32=True
SwitchchkSwitch_32=False
HoldchkHold_32=False
DoublechkDouble_32=False
TripplechkTripple_32=False
Bind type 32=0
Param1_1=28-af-4a-75-d0-01-3c-db
Param2_1=
Param3_1=
Param4_1=
Param5_1=
Param6_1=
Param7_1=
Param1_2=28-de-f4-75-d0-01-3c-99
Param2_2=
Param3_2=
Param4_2=
Param5_2=
Param6_2=
Param7_2=
Param1_3=28-4a-00-75-d0-01-3c-4d
Param2_3=
Param3_3=
Param4_3=
Param5_3=
Param6_3=
Param7_3=
Param1_4=28-e4-b2-75-d0-01-3c-ce
Param2_4=
Param3_4=
Param4_4=
Param5_4=
Param6_4=
Param7_4=
Param1_5=28-53-e7-75-d0-01-3c-0b
Param2_5=
Param3_5=
Param4_5=
Param5_5=
Param6_5=
Param7_5=
Param1_6=28-27-19-75-d0-01-3c-4b
Param2_6=
Param3_6=
Param4_6=
Param5_6=
Param6_6=
Param7_6=
Param1_7=28-ae-fe-75-d0-01-3c-bc
Param2_7=
Param3_7=
Param4_7=
Param5_7=
Param6_7=
Param7_7=
Param1_8=28-fe-20-75-d0-01-3c-6e
Param2_8=
Param3_8=
Param4_8=
Param5_8=
Param6_8=
Param7_8=
Param1_9=
Param2_9=
Param3_9=
Param4_9=
Param5_9=
Param6_9=
Param7_9=
Param1_10=
Param2_10=
Param3_10=
Param4_10=
Param5_10=
Param6_10=
Param7_10=
Param1_11=
Param2_11=
Param3_11=
Param4_11=
Param5_11=
Param6_11=
Param7_11=
Param1_12=
Param2_12=
Param3_12=
Param4_12=
Param5_12=
Param6_12=
Param7_12=
Param1_13=
Param2_13=
Param3_13=
Param4_13=
Param5_13=
Param6_13=
Param7_13=
Param1_14=
Param2_14=
Param3_14=
Param4_14=
Param5_14=
Param6_14=
Param7_14=
Param1_15=
Param2_15=
Param3_15=
Param4_15=
Param5_15=
Param6_15=
Param7_15=
Param1_16=
Param2_16=
Param3_16=
Param4_16=
Param5_16=
Param6_16=
Param7_16=

[Expert]
ManufacturerIdEn=False
ManufacturerId=64447
ManufacturerNameEn=False
ManufacturerName=ptvo.info
ModelIdEn=False
ModelId=ptvo.switch
Icon enabled=False
Icon=
LocationDescriptionEn=False
LocationDescription=
LinkKeyEn=False
LinkKey=01030507090B0D0F00020406080A0C0D
ReportIntEn=True
ReportInt=10
Disable report interval config=False
DateCode=False
WatchDog=True
PSM mode wait commands=False
PSM mode net poll enabled=False
PSM mode net poll interval=0
Disable reset=False

[PA]
TX EN=
RX EN=

[Expert.SBL]
Enabled=False
Pin=

[Expert.OTA]
MOSI=
MISO=
CLK=
CS=
```

However, I don't get any values:

```
{"type":"readResponse","data":{},"linkquality":51,"groupID":0,"cluster":"genOnOff","meta":{"zclTransactionSequenceNumber":216,"manufacturerCode":null,"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0}},"endpoint_id":1}
```

If I don't specify a ds18b20 id, then I only get a temperature value:

```
{"type":"attributeReport","data":{"presentValue":22.75,"description":"C,28-2B-16-75-D0-01-3C-59"},"linkquality":47,"groupID":0,"cluster":"genAnalogInput","meta":{"zclTransactionSequenceNumber":139,"manufacturerCode":null,"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0}},"endpoint_id":1}
```
It should all be listed here, right?

Did I miss something?

Thanks for the great project and greetings
mark

_Originally posted by @MarkStephan89 in https://github.com/ptvoinfo/zigbee-configurable-firmware/discussions/246_

ptvoinfo, 2023-11-09:
@MarkStephan89 You should get a separate similar "attributeReport" for each of the sensors. I've checked your configuratio with two DS18B20 and it works as expected. Could you try with two sensors too?

MarkStephan89, 2023-11-10:
Hi @ptvoinfo 
if I only connect one sensor then it works. :)

I will test it again with several sensors.

Question: can I use the internal resistor or do I absolutely need the external resistor?

Thanks!

MarkStephan89, 2023-11-11:
it is working :)
I reflash the firmware and now it is working :)

Question: 
I have 10 DS18B20. 
The limit is 8 sensors per pin. 

Can I use 2 pins and 5 x ds18b20 on each pin in the free version?

Many thanks!

ptvoinfo, 2023-11-11:
No, it is the total limit of sensors (in the Premium version too). You'll need to make two devices and split sensors between them.

closed 2023-11-11 by MarkStephan89