ptvoinfo/zigbee-configurable-firmware #249: CC2530 UART does not Receive signal but can send

JonRobert, 2023-11-19:
CC2530 UART does not Receive signal but can send.  This seems similar to #236 but I have no hardware issues (see note 1)

I'm using a Hubitat Hub, I realize this community is mostly for MQTT however I believe the UART function is not system dependent.

My PTVO configuration:
[image]

**Notes:** 

1. I have a TI SampleLight firmware that has a UART function and I have used this to prove my hardware is capable of sending data in both directions.  I can't use this except for testing as I cannot change any of the functions.
2. I'm waiting to receive a Zigbee sniffer, not sure how to use it yet or if it can help.
3. I'm connecting the CC2530 board to my PC using a FTDI adapter (in the 3.3v mode)  It has LED's for receive and transmit.  When expecting to receive data from the hub, the FTDI receive LED does not blink.
4. I'm using CoolTerm to send messages from the CC2530 to the hub (it can send hex).
5. I've tried CoolTerm, Arduino and Putty trying to receive message from the CC2530.

ptvoinfo, 2023-11-20:
Are you sure that your hub can send data to CC2530 in the format that it expects? For example, Zigbee2MQTT uses a special converter to send data. The sniffer can definitely help here because we can see a Zigbee data packet and its structure.

Now, I can only offer to try Zigbee2MQTT.

JonRobert, 2023-11-20:
As far as I know I am sending it as ASCII bytes (or at least 8 bit bytes).   Is there a description of the expected format?
I see the different converters, however at this point in time I don't know how to use them.   I'll assume they need to be run in either a Python or javascript environment.

I can complete my current project by using both the UART (external to Hub) and GPIO for Hub to external.

I thank you very much for your support and your excellent firmware generator.
Regards
John

closed 2023-11-20 by JonRobert