ptvoinfo/zigbee-configurable-firmware #252: Ina3221. help

kuradi, 2023-11-27:
hello,
i succesfully got working ina219 with default settings, but when im trying ina3221 i cant get it to work. looks like l2C not communicating at all... is there anybody with successful setting for this sensor?

kuradi, 2023-11-28:
ok. i got it working. finally. but questions remains: how to read 3 channels separately? i have "channel: all" and then it shows quickly all channels in one reading. help?

kuradi, 2023-12-02:
ok i gave up on using the ina3221 because it has a small led on it which makes it pointless for my battery powered project.
now i will try ina 226. and i have problems again.
1. ina 219 immediately gave all readings - V, A, etc. however, ina 226 only offers V.
2. how to connect several in 226 sensors. I currently have 2 sensors - the address of one is 40 (64) and the other is 41 (65) (i checked addresses with esphome and all worked perfectly). if i read https://ptvo.info/zigbee-configurable-firmware-features/external-sensors/i2c-sensor/ i should use the SCL pin once and then assign each ina226 sensor its own SDA pin. Right? or they should also share SDA pin?

ptvoinfo, 2023-12-02:
> ok. i got it working. finally. but questions remains: how to read 3 channels separately? i have "channel: all" and then it shows quickly all channels in one reading. help?

Thank you for your patience! Here is the example for 3 channels.
[image]

ptvoinfo, 2023-12-02:
Of course, it is better to create a custom converter for z2m on the Expert tab.