ptvoinfo/zigbee-configurable-firmware #253: E-Byte E72 interview fails with zigbee2mqtt

u20p17, 2023-12-03:
Hello,
I have a E-Byte E72 want to flash a ptvo firmware with the following settings:
[image]
[image]
[image]

I flashed it with openocd via jtag and tried to pair it with zigbee2mqtt. There the sensor gets recognized, but the interview fails, why?

[image]

u20p17, 2023-12-04:
after setting the following specific settings it seems to work (at least when I change the state of the GPIO input it sends a action message- not sure if it does not send periodic messages because it is configures as end device?):
[image]

seems that the legacy parameter must be set to false... (default is true)...

u20p17, 2023-12-11:
@ptvoinfo : any idea why it does not send mqtt messages via zigbee2mqtt in regular intervals as specified in the Expert settings tab of ptvo?

ptvoinfo, 2023-12-11:
@u20p17 The firmware does not send periodic reports for inputs and UART because the firmware waits for an event from the corresponding pin. For example, if you add a GPIO output, you will receive periodic reports for it.

closed 2024-01-12 by ptvoinfo