ptvoinfo/zigbee-configurable-firmware #254: Intermittent Incorrect Data Reporting from UartSensors

xxddff, 2023-12-08:
Hello,
I would like to express my gratitude for the excellent configurable firmware you have provided. It has been very useful in my projects.

However, I have encountered an issue related to the UartSensor while using the ptvo firmware. I have defined a total of 7 UartSensors, ranging from L2 to L8.

Upon monitoring the data reporting in Home Assistant and Zigbee2Mqtt, I have noticed some irregularities. The sensors seem to be reporting incorrect data continuously. For instance, L3, which is a humidity sensor, is reporting temperature data (which should be reported by L2). Similarly, L4, which is a pressure sensor, is reporting humidity data (which should be reported by L3). This pattern continues for all sensors, except for L2 which seems to be functioning correctly.
(Temperature fluctuations are normal)
[image]
[image]
[image]

Upon reviewing the logs from Zigbee2Mqtt, it appears that the reported endpoint might be incorrect. For example, `{"description":"C","presentValue":18.149999618530273}' from endpoint 3 with groupID 0`, this is L2 data, but the reported endpoint is 3, not 2

Here are the relevant log entries:
```
info  2023-12-08 17:09:30: MQTT publish: topic 'zigbee2mqtt/0x00124b002561df3b', payload '{"action":null,"battery":54.1,"battery_voltage":3.84,"contact_l8":false,"l2":18.15,"l3":65.39,"l4":1006.014,"l5":54,"l6":54.098,"l7":3.836,"linkquality":255,"pressure_l4":1006,"pressure_l5":1005.9,"pressure_l6":1006.2,"pressure_l7":1006.3,"temperature_l2":18.1,"temperature_l3":18.2,"temperature_l4":18.3,"temperature_l5":15.1,"voltage_l7":3.8}'
debug 2023-12-08 17:09:30: Received Zigbee message from '0x00124b002561df3b', type 'attributeReport', cluster 'genAnalogInput', data '{"description":"C","presentValue":18.149999618530273}' from endpoint 3 with groupID 0
info  2023-12-08 17:09:30: MQTT publish: topic 'zigbee2mqtt/0x00124b002561df3b', payload '{"action":null,"battery":54.1,"battery_voltage":3.84,"contact_l8":false,"l2":18.15,"l3":18.15,"l4":1006.014,"l5":54,"l6":54.098,"l7":3.836,"linkquality":255,"pressure_l4":1006,"pressure_l5":1005.9,"pressure_l6":1006.2,"pressure_l7":1006.3,"temperature_l2":18.1,"temperature_l3":18.1,"temperature_l4":18.3,"temperature_l5":15.1,"voltage_l7":3.8}'
debug 2023-12-08 17:09:30: Received Zigbee message from '0x00124b002561df3b', type 'attributeReport', cluster 'genAnalogInput', data '{"presentValue":65.3896484375}' from endpoint 4 with groupID 0
info  2023-12-08 17:09:30: MQTT publish: topic 'zigbee2mqtt/0x00124b002561df3b', payload '{"action":null,"battery":54.1,"battery_voltage":3.84,"contact_l8":false,"l2":18.15,"l3":18.15,"l4":65.39,"l5":54,"l6":54.098,"l7":3.836,"linkquality":255,"pressure_l4":1006,"pressure_l5":1005.9,"pressure_l6":1006.2,"pressure_l7":1006.3,"temperature_l2":18.1,"temperature_l3":18.1,"temperature_l4":18.3,"temperature_l5":15.1,"voltage_l7":3.8}'
debug 2023-12-08 17:09:30: Received Zigbee message from '0x00124b002561df3b', type 'attributeReport', cluster 'genAnalogInput', data '{"description":"Pa","presentValue":1006.0072021484375}' from endpoint 5 with groupID 0
info  2023-12-08 17:09:30: MQTT publish: topic 'zigbee2mqtt/0x00124b002561df3b', payload '{"action":null,"battery":54.1,"battery_voltage":3.84,"contact_l8":false,"l2":18.15,"l3":18.15,"l4":65.39,"l5":1006.007,"l6":54.098,"l7":3.836,"linkquality":255,"pressure_l4":1006,"pressure_l5":1006,"pressure_l6":1006.2,"pressure_l7":1006.3,"temperature_l2":18.1,"temperature_l3":18.1,"temperature_l4":18.3,"temperature_l5":15.1,"voltage_l7":3.8}'
debug 2023-12-08 17:09:30: Received Zigbee message from '0x00124b002561df3b', type 'attributeReport', cluster 'genAnalogInput', data '{"presentValue":54}' from endpoint 6 with groupID 0
info  2023-12-08 17:09:30: MQTT publish: topic 'zigbee2mqtt/0x00124b002561df3b', payload '{"action":null,"battery":54,"battery_voltage":3.84,"contact_l8":false,"l2":18.15,"l3":18.15,"l4":65.39,"l5":1006.007,"l6":54,"l7":3.836,"linkquality":255,"pressure_l4":1006,"pressure_l5":1006,"pressure_l6":1006.2,"pressure_l7":1006.3,"temperature_l2":18.1,"temperature_l3":18.1,"temperature_l4":18.3,"temperature_l5":15.1,"voltage_l7":3.8}'
debug 2023-12-08 17:09:30: Received Zigbee message from '0x00124b002561df3b', type 'attributeReport', cluster 'genAnalogInput', data '{"presentValue":54.09765625}' from endpoint 7 with groupID 0
info  2023-12-08 17:09:30: MQTT publish: topic 'zigbee2mqtt/0x00124b002561df3b', payload '{"action":null,"battery":54,"battery_voltage":54.1,"contact_l8":false,"l2":18.15,"l3":18.15,"l4":65.39,"l5":1006.007,"l6":54,"l7":54.098,"linkquality":255,"pressure_l4":1006,"pressure_l5":1006,"pressure_l6":1006.2,"pressure_l7":1006.3,"temperature_l2":18.1,"temperature_l3":18.1,"temperature_l4":18.3,"temperature_l5":15.1,"voltage_l7":3.8}'
debug 2023-12-08 17:09:30: Received Zigbee message from '0x00124b002561df3b', type 'attributeReport', cluster 'genAnalogInput', data '{"description":"V","presentValue":3.836250066757202}' from endpoint 7 with groupID 0
info  2023-12-08 17:09:30: MQTT publish: topic 'zigbee2mqtt/0x00124b002561df3b', payload '{"action":null,"battery":54,"battery_voltage":3.84,"contact_l8":false,"l2":18.15,"l3":18.15,"l4":65.39,"l5":1006.007,"l6":54,"l7":3.836,"linkquality":255,"pressure_l4":1006,"pressure_l5":1006,"pressure_l6":1006.2,"pressure_l7":1006.3,"temperature_l2":18.1,"temperature_l3":18.1,"temperature_l4":18.3,"temperature_l5":15.1,"voltage_l7":3.8}'
```

Here is my firmware configuration:

```
Board type: CC2530 + RFX2401 (configure PA control pins!!!)
Device type: End device without routing
 Transmit enable (TXEN): P11, Receive enable (RXEN): P10
Manufacturer name: home
Model ID: home.uart
Update firmware's timestamp : 2023-12-06
Set default reporting interval (s): 60

Output pins:
P02: Output 1, UART (UART: 115200 8N1, Packet end: None, Byte: 239)
P35: Output 2, UART sensor (Mode: Analog value, : Reporting, Units: Temperature (C)), Remember state
P36: Output 3, UART sensor (Mode: Analog value, : Reporting, Units: Humidity (%)), Remember state
P37: Output 4, UART sensor (Mode: Analog value, : Reporting, Units: Pressure (Pa)), Remember state
P34: Output 5, UART sensor (Mode: Analog value, : Reporting, Units: None), Remember state
P32: Output 6, UART sensor (Mode: Analog value, : Reporting, Units: None), Remember state
P33: Output 7, UART sensor (Mode: Analog value, : Reporting, Units: Voltage (V)), Remember state
P31: Output 8, UART sensor (Mode: On/Off, : Reporting, Units: None), Remember state
```

I am using stm32f103c8t6 as the UART sensor and Arduino for programming. Here is the relevant Arduino code:

```
#include <Wire.h>
#include <Adafruit_Sensor.h>
#include <Adafruit_BME280.h>
#include <Adafruit_SGP40.h>

// ptvo
#include "DataHandler.h" 
#include "DataDispatcherBase.h"
#include "extensions/PtvoUartSensor.h"

#include <SparkFun_MAX1704x_Fuel_Gauge_Arduino_Library.h>

SFE_MAX1704X lipo; // Defaults to the MAX17043

// #define SDA_PIN 26
[... 110 lines cut ...]
    // lipo.getVoltage() returns a voltage value (e.g. 3.93)
    Voltage = lipo.getVoltage();
    // lipo.getSOC() returns the estimated state of charge (e.g. 79%)
    Percentage = lipo.getSOC();
    // lipo.getAlert() returns a 0 or 1 (0=alert not triggered)
    Alert = lipo.getAlert();

    SerialCommandHandler.SendVariable(ENDPOINT_Temperature);
    SerialCommandHandler.SendVariable(ENDPOINT_Humidity);
    SerialCommandHandler.SendVariable(ENDPOINT_Pressure);
    if (TVOC_index <= 500) {
      SerialCommandHandler.SendVariable(ENDPOINT_TVOC);
    }
    SerialCommandHandler.SendVariable(ENDPOINT_Percentage);
    SerialCommandHandler.SendVariable(ENDPOINT_Voltage);
    SerialCommandHandler.SendVariable(ENDPOINT_Alert);
    
    lastSentTime = millis();
  }

  delay(1000);
}
```

I would appreciate any assistance you could provide in resolving this issue. Thank you for your time and consideration.

ptvoinfo, 2023-12-12:
@xxddff Could you please try the latest version?
https://ptvo.info/download/ptvo-firmware-latest.zip?20231212

xxddff, 2023-12-12:
@ptvoinfo Thank you for your prompt response and for providing the latest version. I have installed it and so far it is working fine. I will monitor the data for one more day and if there are no issues.

closed 2023-12-13 by xxddff