ptvoinfo/zigbee-configurable-firmware #255: Device unsupported

jsv93, 2023-12-18:
I am having trouble getting devices to become supported via external converters. I have tried uninstalling z2m and starting again but have not had any luck. Whats interesting is that a device which ahd previously woreked with its firmware and external converter no longer works. 

I have tried adding a basic setup, only 1 GPIO output and I get the same behaviour. The device interviews and joins but does not become supported even after adding the converter. I am using the converter directly from the firmware configurator with no success.

Any suggestions?
[image]
[image]
[image]
ns?

ptvoinfo, 2023-12-19:
@jsv93 Hi,

Did you add your converter to configuration.yaml?
https://www.zigbee2mqtt.io/guide/configuration/more-config-options.html#external-converters

jsv93, 2023-12-19:
Yes, I tried adding via the front end interface and manually via the configuration.yaml file. Both do not work. 

I have just tried resetting the firmware configurator to defaults with only 1 gpio output and without changing the modelID (ptvo.switch). Meaning no external converter is needed. This passes the interview and is immediately supported.

ptvoinfo, 2023-12-19:
If Z2M shows "unsupported", it cannot find a converter for the corresponding ModelID.
You can enable debug logs in z2m and make additional tests. Maybe, Z2M cannot load the converter.

jsv93, 2023-12-19:
It appears as this is the case. However I am having trouble figuring out why this is happening. Nothing changed since the system was previously working. It appears as this may not actually be an issue at all. Ill report back when I find a solution.
Thanks for your help.

jsv93, 2023-12-20:
After some further research and playing with z2m I finaly managed to get an error in the log which indicates where the issue is. However I am not exactly sure how to interpret the error. The advise from z2m support suggested that the issue was with the converter itself. The error is:

Failed to load external converter file 'Bed_Pres_Multi.js' (Cannot read properties of undefined (reading 'access'))

Any idea what this may be indicating?

jsv93, 2023-12-20:
After a bit of research and checking in with the support page of z2m it appears the converters being generated by the ptvo firmware configurator are currently incorrect.

Replacing the inital part of the converter from the configurator generated external converter:

```
const zigbeeHerdsmanConverters = require('zigbee-herdsman-converters');
const zigbeeHerdsmanUtils = require('zigbee-herdsman-converters/lib/utils');

const exposes = zigbeeHerdsmanConverters.exposes;
const ea = exposes.access;
const e = exposes.presets;
const fz = zigbeeHerdsmanConverters.fromZigbeeConverters;
const tz = zigbeeHerdsmanConverters.toZigbeeConverters;
```

With:

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const extend = require('zigbee-herdsman-converters/lib/extend');
const e = exposes.presets;
const ea = exposes.access;
```

Immediately resolved the issue. It appears as though the configuration for defining values in the external converters has changed vs what is being generated via the software configurator. 

Perhaps this needs to be update or at least made available as a step when creating external converters.

ptvoinfo, 2023-12-20:
Recreate this converter file in the latest version of the configurator.

closed 2024-01-01 by ptvoinfo