ptvoinfo/zigbee-configurable-firmware #258: Tiny bug in generated custom ZHA device handler

adamantivm, 2024-01-11:
I noticed that the ZHA handlers saved with the configurator have some spaces after the manufacturer name, which causes ZHA to fail to load the device handler (quirk).
Removing those extra spaces solves the problem as a workaround.

I'm using PTVO. Firmware Configuration 1.9.0.11 (Firmware: 2023-12-12).
I've never changed the deafult manufacturer name, left it default, unchanged.

Here is an excerpt from such a generated quirk:

```
...

    signature = {
        MODELS_INFO: [("ptvo.info       ", PTVO_MODEL_ID)],
        ENDPOINTS: {
...
```

It's a minor annoyance, I thought I'd report it as it may be very easy to fix.

adamantivm, 2024-01-11:
NOTE: If I change the default manufacturer name, the problem doesn't manifest itself, making this even less of an issue.

I would still suggest making it work well with the default value, as otherwise this adds another point of error for first time users (defaults don't work)

ptvoinfo, 2024-01-12:
Could you please try the latest version of the configuration from the main web site?
https://ptvo.info/download/ptvo-firmware-latest.zip?1

adamantivm, 2024-01-17:
The updated worked @ptvoinfo.
Quirks produced with the default manufacturer name using 1.9.0.14 ( 2023-12-12) don't include the extra spaces.
Thanks for the quick update.

closed 2024-01-17 by adamantivm