ptvoinfo/zigbee-configurable-firmware #260: DS18B20 on 1.90.13

mipoat, 2024-01-14:
Hi,

I've got troubles to get one connected DS18B20 up and running with 1.9.0.13 (Firmware 2023-12-12).
I've generated the firmware with this config:
`Board type: CC2530
Device type: End device without routing

Output pins:
P15: Output 1, DS18B20, External pull-up (Sensor ID (optional, hex): , Mode: Single, Auto-detect)
P17: Output 2, GPIO, Pull-up (Role: Generic), Remember state`

I can pair the device with my Zigbee2MQTT running on a raspberryPi.
I can control P17 via the toggle switch in the GUI of Zigbee2MQTT, so everything is up and running.
However I don't see anywhere the temperature values. Using an oscilloscope, I can see 1 wire communication with the DS18B20 every minute. Signal is looking perfect, I'm using an external pull-up.
I've not gone down the road and tried to decode the 1 wire protocol, for the time being that is too cumbersome for me.
Using a totally different DS18B20 does not help either. Both DS18B20, from a totally different source/brand, are working fine on an Arduino board. 

I'm running out of ideas, anything I could try?

this is what I get in the Zigbee2MQTT logs:
Info 2024-01-14 11:04:31MQTT publish: topic 'zigbee2mqtt/0x00124b0013e7f3a3', payload '{"l1":null,"linkquality":83,"state_l1":"OFF","state_l2":"ON"}'
Info 2024-01-14 11:05:31MQTT publish: topic 'zigbee2mqtt/0x00124b0013e7f3a3', payload '{"l1":null,"linkquality":87,"state_l1":"OFF","state_l2":"ON"}'
Info 2024-01-14 11:06:32MQTT publish: topic 'zigbee2mqtt/0x00124b0013e7f3a3', payload '{"l1":null,"linkquality":94,"state_l1":"OFF","state_l2":"ON"}'
Info 2024-01-14 11:07:31MQTT publish: topic 'zigbee2mqtt/0x00124b0013e7f3a3', payload '{"l1":null,"linkquality":98,"state_l1":"OFF","state_l2":"ON"}'
Info 2024-01-14 11:08:31MQTT publish: topic 'zigbee2mqtt/0x00124b0013e7f3a3', payload '{"l1":null,"linkquality":98,"state_l1":"OFF","state_l2":"ON"}'
Info 2024-01-14 11:09:31MQTT publish: topic 'zigbee2mqtt/0x00124b0013e7f3a3', payload '{"l1":null,"linkquality":98,"state_l1":"OFF","state_l2":"ON"}'

ptvoinfo, 2024-01-14:
Please enable debug logs and check incoming Zigbee data.

mipoat, 2024-01-14:
Interesting, I do see the temperature values in debug log:
Debug 2024-01-14 11:46:17Received Zigbee message from '0x00124b0013e7f3a3', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2362}' from endpoint 1 with groupID 0
Debug 2024-01-14 11:47:16Received Zigbee message from '0x00124b0013e7f3a3', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 2 with groupID 0
Info 2024-01-14 11:47:16MQTT publish: topic 'zigbee2mqtt/0x00124b0013e7f3a3', payload '{"l1":null,"linkquality":98,"state_l1":"OFF","state_l2":"ON"}'
Debug 2024-01-14 11:47:17Received Zigbee message from '0x00124b0013e7f3a3', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2368}' from endpoint 1 with groupID 0
Debug 2024-01-14 11:48:16Received Zigbee message from '0x00124b0013e7f3a3', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 2 with groupID 0
Info 2024-01-14 11:48:16MQTT publish: topic 'zigbee2mqtt/0x00124b0013e7f3a3', payload '{"l1":null,"linkquality":76,"state_l1":"OFF","state_l2":"ON"}'
Debug 2024-01-14 11:48:17Received Zigbee message from '0x00124b0013e7f3a3', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":2368}' from endpoint 1 with groupID 0
Debug 2024-01-14 11:49:16Received Zigbee message from '0x00124b0013e7f3a3', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 2 with groupID 0
Info 2024-01-14 11:49:16MQTT publish: topic 'zigbee2mqtt/0x00124b0013e7f3a3', payload '{"l1":null,"linkquality":98,"state_l1":"OFF","state_l2":"ON"}'
Debug 2024-01-14 11:49:17Received Zigbee message from '0x00124b0013e7f3a3', type 'attributeReport', cluster 'msTemperatureMeasurement', data '{"measuredValue":3012}' from endpoint 1 with groupID 0

But it is nowhere presented in the GUI of Zigbee2MQTT. Is this normal behaviour?

Other temp sensors are reporting the value in the payload:
Debug 2024-01-14 11:49:13Received Zigbee message from 'TempSonoffOffice', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":5811}' from endpoint 1 with groupID 0
Info 2024-01-14 11:49:13MQTT publish: topic 'zigbee2mqtt/TempSonoffOffice', payload '{"battery":100,"humidity":54.11,"linkquality":80,"temperature":22.57,"voltage":3100}'
Debug 2024-01-14 11:49:13Received Zigbee message from 'TempSonoffOffice', type 'attributeReport', cluster 'msRelativeHumidity', data '{"measuredValue":5811}' from endpoint 1 with groupID 0
Info 2024-01-14 11:49:13MQTT publish: topic 'zigbee2mqtt/TempSonoffOffice', payload '{"battery":100,"humidity":54.11,"linkquality":80,"temperature":22.57,"voltage":3100}'

mipoat, 2024-01-14:
all good, this issue can be closed

after I saw the temperature value in the debug message, I figured this is not a device but rather a zigbe2mqtt issue.
I updated zigbe2mqtt and now I get the temp value in the payload:

Info 2024-01-14 14:32:13MQTT publish: topic 'zigbee2mqtt/0x00124b0013e7f3a3', payload '{"l1":null,"linkquality":72,"state_l1":"OFF","state_l2":"ON","temperature_l1":23.37}'

closed 2024-01-15 by ptvoinfo