ptvoinfo/zigbee-configurable-firmware #261: no external sensor voltage

mipoat, 2024-01-16:
I'm having a hard time getting supply voltage for my DS18B20 sensors when P10 is configured as external sensor power control.

I have a premium license and I'm generating a PSM hex file.
I've hooked up 2 DS18B20 sensors to VCC and can read them out without any problem with the configuration below.
However, if I use P10 as supply voltage for these sensors, I'm not getting any data.
Measuring P10 with an oscilloscope I don't see any signal, only 0V no matter for how long I inspect this pin.
This is true even when nothing is attached to P10 and it is floating without any load attached to it.

my config:
[image]

[image]

If I generate an end device firmware and set P10 as GPIO output, I can toggle it with the toggle switch in the zigbee2mqtt GUI.
So the pin is working.

Any ideas?

mipoat, 2024-01-17:
Out of despair I tried the exact same version like here: https://github.com/ptvoinfo/zigbee-configurable-firmware/issues/159

Same configuration options produce 3,3V on P10 with version 2022-10-20 everytime the device wakes up and reads the sensors:
[image]

Can you @ptvoinfo please check what is wrong with the latest version?
I'm not sure if it is a good idea to use the old version 2022-10-20, because I see some DS18B20 related fixes in the changelog between this and the most recent version.

keynote11, 2024-02-25:
I have the same issue with a cc2530.

ptvoinfo, 2024-03-02:
@keynote11 Did you try the latest version of the firmware?

keynote11, 2024-03-04:
@ptvoinfo yes try with the latest firmware. This is my config:

[image]

ptvoinfo, 2024-03-19:
@keynote11 Could you please try the latest version?

keynote11, 2024-03-20:
@ptvoinfo if the sensor is sending, i have on p10 3 Volts, is this correct?

ptvoinfo, 2024-03-21:
@keynote11 yes, correct.

closed 2024-03-31 by ptvoinfo