ptvoinfo/zigbee-configurable-firmware #266: Pressure data lost from SHT30

ch00kan0ff, 2024-02-28:
Hi!
I make firmware with 2 sensors SHT30 and ds18b20

`
Board type: CC2530
Device type: Router

Output pins:
P06: Output 1, Temperature sensor (I2C), Pull-up (Address (dec): 68, Type: SHT30)
P04: Output 2, I2C SCL/SPI CLK, Pull-up
P02: Output 3, DS18B20, External pull-up (Sensor ID (optional, hex): 28-53-53-81-E3-15-3C-C9, Mode: Multiple on bus, Auto-detect)
`
And pressure from SHT30 don't appear in Z2M. 
[image]

Without DS18B20 all was good.

firmware configurator version 1.9.0.14, Z2M version [1.34.0] commit: [aae7312]

I try to generate external converter, and there is no pressure in exposes:

`
const device = {
    zigbeeModel: ['ch_ds18_sht30'],
    model: 'ch_ds18_sht30',
    vendor: 'Custom devices (DiY)',
    description: '[Configurable firmware](https://ptvo.info/zigbee-configurable-firmware-features/)',
    fromZigbee: [fz.ignore_basic_report, fz.temperature, fz.ptvo_humidity, fz.ptvo_pressure, fz.ptvo_switch_analog_input,],
    toZigbee: [tz.ptvo_switch_trigger,],
    exposes: [e.temperature().withEndpoint('l1'),
      e.humidity().withEndpoint('l1'),
      e.temperature().withEndpoint('l3'),
],
    meta: {
        multiEndpoint: true,
        
    },
    endpoint: (device) => {
        return {
            l1: 1, l3: 3,
        };
    },
    configure: async (device, coordinatorEndpoint, logger) => {
            const endpoint = device.getEndpoint(1);
      await endpoint.read('genBasic', ['modelId', 'swBuildId', 'powerSource']);
    },

};

module.exports = device;
`
How can I get pressure back? 
`Add e.pressure().withEndpoint('l1')` 
to exposes in converter? And flash the device with new firmware whith custom model name to z2m uses external converter?

ch00kan0ff, 2024-03-01:
Clusters in Z2M looks like this:
[image]

ptvoinfo, 2024-03-02:
@ch00kan0ff Hmm. I cannot find the SHT30 datasheet that it can return a pressure value. Temperature & Humidity only.

ch00kan0ff, 2024-03-02:
Oh. I am really sorry. My first version was with bme. And after change it to sht30 I forgot that it does not have pressure. Thank you for answer and for great firmware!

closed 2024-03-02 by ch00kan0ff