ptvoinfo/zigbee-configurable-firmware #268: CC2530/DL-20 no states

derSchreifritz, 2024-03-21:
### What happened?

I flashed a DL-20 CC2530 with custom PVTO-tool config. 
Z2M with conbee2 works fine, with other devices. The CC2530 is paired (when I touch it.. something wrong with the battery power?).
Problem is, no state report, state is {}, not even Led.  
edit: after another pairing, state is: {
    "linkquality": 255
}
no topic in mqtt explorer still.

here is a screenshot of Clusters, there is 6 more endpoints like no.2
[image]
When flashed with the default from z-stack it works.

### What did you expect to happen?

at least LED status is reported.
Trying to connect Dallas sensors.

### Zigbee2MQTT version

1.36.0-1

### Adapter firmware version

6.23.0

### Adapter

deconz/ conbee2

### Setup

add-on ha

this is the log when pairing:
info 2024-03-21 10:12:05Device '0x00124b0026171a69' joined
info 2024-03-21 10:12:05MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b0026171a69","ieee_address":"0x00124b0026171a69"},"type":"device_joined"}'
info 2024-03-21 10:12:05MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0x00124b0026171a69"},"type":"device_connected"}'
info 2024-03-21 10:12:05Starting interview of '0x00124b0026171a69'
info 2024-03-21 10:12:05MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b0026171a69","ieee_address":"0x00124b0026171a69","status":"started"},"type":"device_interview"}'
info 2024-03-21 10:12:05MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00124b0026171a69"},"type":"pairing"}'
info 2024-03-21 10:12:07Successfully interviewed '0x00124b0026171a69', device has successfully been paired
info 2024-03-21 10:12:07Device '0x00124b0026171a69' is supported, identified as: Custom devices (DiY) CC2530 router (CC2530.ROUTER)
info 2024-03-21 10:12:07MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"CC2530 router","exposes":[{"access":1,"label":"Led","name":"led","property":"led","type":"binary","value_off":false,"value_on":true},{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"CC2530.ROUTER","options":[],"supports_ota":false,"vendor":"Custom devices (DiY)"},"friendly_name":"0x00124b0026171a69","ieee_address":"0x00124b0026171a69","status":"successful","supported":true},"type":"device_interview"}'
info 2024-03-21 10:12:07MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"CC2530 router","friendly_name":"0x00124b0026171a69","model":"CC2530.ROUTER","supported":true,"vendor":"Custom devices (DiY)"},"type":"pairing"}'
info 2024-03-21 10:12:07MQTT publish: topic 'homeassistant/binary_sensor/0x00124b0026171a69/led/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00124b0026171a69"],"manufacturer":"Custom devices (DiY)","model":"CC2530 router (CC2530.ROUTER)","name":"0x00124b0026171a69","sw_version":"2021-07-05","via_device":"zigbee2mqtt_bridge_0x00212effff09a813"},"json_attributes_topic":"zigbee2mqtt/0x00124b0026171a69","name":"Led","object_id":"0x00124b0026171a69_led","origin":{"name":"Zigbee2MQTT","sw":"1.36.0","url":"[https://www.zigbee2mqtt.io"},"payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0x00124b0026171a69","unique_id":"0x00124b0026171a69_led_zigbee2mqtt","value_template":"{{](https://www.zigbee2mqtt.io%22%7D,%22payload_off%22:false,%22payload_on%22:true,%22state_topic%22:%22zigbee2mqtt/0x00124b0026171a69%22,%22unique_id%22:%220x00124b0026171a69_led_zigbee2mqtt%22,%22value_template%22:%22%7B%7B) value_json.led }}"}'
info 2024-03-21 10:12:07MQTT publish: topic 'homeassistant/sensor/0x00124b0026171a69/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0x00124b0026171a69"],"manufacturer":"Custom devices (DiY)","model":"CC2530 router (CC2530.ROUTER)","name":"0x00124b0026171a69","sw_version":"2021-07-05","via_device":"zigbee2mqtt_bridge_0x00212effff09a813"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x00124b0026171a69","name":"Linkquality","object_id":"0x00124b0026171a69_linkquality","origin":{"name":"Zigbee2MQTT","sw":"1.36.0","url":"[https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/0x00124b0026171a69","unique_id":"0x00124b0026171a69_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{](https://www.zigbee2mqtt.io%22%7D,%22state_class%22:%22measurement%22,%22state_topic%22:%22zigbee2mqtt/0x00124b0026171a69%22,%22unique_id%22:%220x00124b0026171a69_linkquality_zigbee2mqtt%22,%22unit_of_measurement%22:%22lqi%22,%22value_template%22:%22%7B%7B) value_json.linkquality }}"}'

and this the config hex from pvto:

Board type: CC2530
Device type: End device without routing
Model ID: ptvo.switch1
Status LED: P10 (Inversed), Periodic (every 5 seconds)
Set default reporting interval (s): 20

Output pins:
P03: Output 1, DS18B20, Pull-up (Sensor ID (optional, hex): )
P03: Output 2, DS18B20, External pull-up (Sensor ID (optional, hex): )
P02: Output 3, DS18B20, Pull-up (Sensor ID (optional, hex): )
P03: Output 4, DS18B20, Pull-up (Sensor ID (optional, hex): )
P22: Output 5, DS18B20, Pull-up (Sensor ID (optional, hex): )
P21: Output 8, DS18B20, Pull-up (Sensor ID (optional, hex): )

ptvoinfo, 2024-03-22:
If you would like to control a LED from Z2M, you should remove this pin from the "Status LED" option.
So, define 
P10, Output 1, GPIO, Pull-up, Role "General".
Re-flash the device, delete it in Z2M, restart Z2M, then join the device again.

derSchreifritz, 2024-03-22:
restart Z2M add-on in ha did the job.
Embarrassing, thanks for the help.
Now I see one of the 3 sensors showing up with an ID. Maybe that's a cabeling issue or the faulty sensor.

closed 2024-03-22 by derSchreifritz