ptvoinfo/zigbee-configurable-firmware #274: Ultrasonic distance sensor (UART) implementation is wrong.

bullar, 2024-04-15:
In PTVO FW version (1.9.0.15/2024-04-06) the selection of the different sensor types (e.g. US100,..) doesn't work. 
Independant from the type selection the FW behavior is always the same:
```
TX: 0x55
expected RX: 0x55 0xAA 0x11 0x02 [0x02 0x12] 0x34 0x5A
Distance value: High 8-bit: 0x02, Low 8-bit : 0x12
```

Correct would be for sensor **JSN-sr04t, DYP-A01, A02YYUW**:
```
TX: 0x55 
RX: 0xFF MSB LSB CRC 
 0xFF – fixed header byte. 
 MSB, LSB distance value 
 CRC – Checksum as a sum of all bytes before.
Example: 0xFF 0x00 0xDE 0xDE
```

and sensor **DFRobot URM06, URM07**:
```
TX: 0x55 0xAA 0x11 0x00 0x02 0x12
  Instruction： 
  Header----------------0x55
  Header----------------0xAA
  Address---------------0x11
  Length----------------0x00
  Read Distance---------0x02
  Check Sum-------------0x12

RX: 0x55 0xAA 0x11 0x02 0x02 [0x12 0x34] 0x5A
  High 8-bit: 0x12
  Low 8-bit : 0x34
So the distance value is “0x1234”, which convert to decimal is 4660mm.
“0x5A” is low 8-bit of the check sum.
```

bullar, 2024-04-29:
Great, will test it when I am back home next weekend...

bullar, 2024-05-04:
@ptvoinfo 
**sensor DFRobot URM06, URM07:** 
working well now ✅ (tested with serial emulator, don't have this sensor type available)

**sensor JSN-sr04t, DYP-A01, A02YYUW:**
The expected checksum is wrong ❌
The firmware (2024-04-29) expects [sum of all byte -1] instead of only [sum of all bytes].
`Sensor response:       0xFF 0x00 0xD5 0xD5    --> value 213`
`Firmware expectation:  0xFF 0x00 0xD5 0xD4` 
(tested with serial emulator and real sensor)
[image]

bullar, 2024-05-05:
@ptvoinfo  At least all my 6 sensors of type AJ-S04M show the same response with checksum [MSB+LSB]:
[image]
But you are right the specification defines [sum of all bytes]. In another sensor description I found [MSB+LSB] as checksum calculation (same as sum of all bytes+1). I am not sure if there are two different types in the market...

ptvoinfo, 2024-05-25:
@bullar Or just change the unit name for the sensor :)

closed 2024-06-03 by ptvoinfo