ptvoinfo/zigbee-configurable-firmware #275: UART Sensor level control -> state can't be controlled by external MCU

bullar, 2024-04-18:
I used the provided **UART sensor example**  (1.9.0.15/2024-04-06) including the level control. All sensors are working fine except the state of the light can't be controlled **from** external MCU.

If the light is ON and the level control is set to any value (here 141):
`MQTT publish: topic 'zigbee2mqtt/0x00124b002a510550', payload '{"bind_command_l1":"on/off","brightness_l4":141,"link_to_output_l1":"no","linkquality":49,"state_l1":"OFF","state_l2":"OFF","state_l4":"ON","switch_actions_l1":"on","switch_type_l1":"switch"}'`

the level can be controlled by external MCU with the command sequence:
`<STX> 0x13 0x6D <ETX> 0xDC` to 109 for instance.
or with sequence:
`<STX> 0x13 0x00 <ETX> 0x6F`
 set to 0. 
Here the log:
`MQTT publish: topic 'zigbee2mqtt/0x00124b002a510550', payload '{"bind_command_l1":"on/off","brightness_l4":0,"link_to_output_l1":"no","linkquality":57,"state_l1":"OFF","state_l2":"OFF","state_l4":"ON","switch_actions_l1":"on","switch_type_l1":"switch"}'`

But the state of endpoint 4 stays "ON" (here: `"state_l4":"ON"`). 
I didn't find a way to set it OFF from external MCU even with boolean data format (`<STX> 0x03 0x00 <ETX> 0x5F`')
The same issue when trying to set the state from OFF to ON.

Btw, if controlling the state from ZIGBEE2MQTT GUI the following commands are send **to** external MCU:
Set ON: `<STX> 0x13 0x6D <ETX> 0xDC`
Set OFF: `<STX> 0x13 0x00 <ETX> 0x6F` 

**Does anyone have an idea what's wrong or is it a bug?**

ptvoinfo, 2024-04-29:
@bullar Could you please check the latest version where I've tried to resolve this issue?

bullar, 2024-05-05:
Fixed in firmware 2024-04-29 ✅🙏

closed 2024-05-23 by ptvoinfo