ptvoinfo/zigbee-configurable-firmware #276: router firmware error

flyice-juin, 2024-04-18:
Hello, The situation is like this. For the cc2530 chip, it is normal to use "end device without routing", but if it use "router", it cannot be paired with zigbee2mqtt. It is directly unable to pair, not not supported.
Tested 2 chips, including CC2530F256 and E18-M1S-PCB, both are like this

bullar, 2024-04-18:
I guess it's not a general problem. The following router setup is running without issues as watermeter on my side:

[image]

But it's paired with ZHA instead of ZIGBEE2MQTT.
Please share more details about your configuration...

flyice-juin, 2024-04-21:
Change the APP to the new version , it is ok.
I want to use the psm , but there is no way pay for online in china , is there other way buy it ?

ptvoinfo, 2024-04-21:
@flyice-juin You can purchase in China if you can pay using AliPay.

flyice-juin, 2024-04-21:
> @flyice-juin You can purchase in China if you can pay using AliPay.

Indeed, it can be paid, but the price is too expensive for both individuals and businesses. Is it possible to consider significantly lowering the price?

ptvoinfo, 2024-04-21:
@flyice-juin What do you mean by "significantly"? Please note, the payment processing operator costs 2 USD for each order.

closed 2024-05-23 by ptvoinfo