ptvoinfo/zigbee-configurable-firmware #277: Led firmware control only white channel

Sennevds, 2024-04-22:
I've bought the premium features add created following config:
[image]
It's for a 24v led strip which has 4 separate inputs (RGBW) This works and I can perfectly control the color and brightness.
I can't find how to set the PWM to 0 on the RGB channels and only control the white channel.
Is this possible?

ptvoinfo, 2024-04-23:
@Sennevds 
Did you configure your firmware in FirmwareConfigLED.exe?

Sennevds, 2024-04-23:
Yes, I've found out that if I send #ffffff all color leds go out but for some reason the white leds won't turn on I will check my pcb on errors this evening.
Still strange that only when sending #ffffff it will turn on the white leds I would think that there would be a toggle in Home Assistant between rgb and white

ptvoinfo, 2024-04-23:
@Sennevds It is by design. The white channel is active only when colors are inactive. So, you can color or white modes.

Sennevds, 2024-04-23:
Okay thanks for the info, probably just a weird way Home Assistant works with rgbw lights.
Because I need to manually set the color to white it's not possible to easily toggle between rgb and white.

Sennevds, 2024-04-25:
@ptvoinfo I've created a test setup (with the none led firmware because other device and not having a premium for that one).
The PWM signal is working great for 3 pin's (P16, P20 and P17) but it's not working on pin P23.
This is my config:
[image]
[pool.txt](https://github.com/ptvoinfo/zigbee-configurable-firmware/files/15119520/pool.txt)

ptvoinfo, 2024-04-29:
As I see, P23 is used for an external oscillator. Or did you assemble a custom board without it?

ptvoinfo, 2024-04-29:
I cannot open this page (product not found). The most boards have 32 kHz oscillators. Check your device.

Sennevds, 2024-05-01:
Checked pinout and was indeed connected to oscillator. 
Last thing about the rgbw configuration. The light is recognized as a RGB light and you automatically switch to the white LEDs when you select white color. Home assistant has a option to identify a light as rgbw and then has a different control for the white LEDs. Is this possible?