ptvoinfo/zigbee-configurable-firmware #281: INA219

rickbhs, 2024-06-01:
Hello, 
I have a problem with the implementation von a INA219 with CC2530. I cant get it work. 
- BME280 sensors are working
- other I2C are working
- I tested multiple INA219 chips
- Tested multiple CC2530 chips 
- tested PTVO Premium 

But all the times shows Zigbee2MQTT. N/A at Voltage, current and power

Can some one point me in the right direction?

ptvoinfo, 2024-06-03:
@rickbhs Hello, 
Are you using the latest version of Z2M? Did you try to create a custom converter for your device? D
Can you enable debug logs in Z2M and check raw incoming Zigbee message?

rickbhs, 2024-06-03:
Version Zigbee2MQTT 1.38.0-dev and without dev
Yes with a custom converter and without

Some logs:
[2024-06-03 19:31:06] debug: 	zh:zstack:unpi:parser: --> parsed 27 - 2 - 4 - 129 - [0,0,6,0,171,117,3,1,0,99,0,213,208,14,0,0,7,8,42,10,0,0,16,0,146,186,27] - 96
[2024-06-03 19:31:06] debug: 	zh:zstack:znp: AREQ: <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":30123,"srcendpoint":3,"dstendpoint":1,"wasbroadcast":0,"linkquality":99,"securityuse":0,"timestamp":970965,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[8,42,10,0,0,16,0]}}
[2024-06-03 19:31:06] debug: 	zh:controller: Received payload: clusterID=6, address=30123, groupID=0, endpoint=3, destinationEndpoint=1, wasBroadcast=false, linkQuality=99, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"manufacturerCode":null,"transactionSequenceNumber":42,"commandIdentifier":10},"payload":[{"attrId":0,"dataType":16,"attrData":0}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
[2024-06-03 19:31:06] debug: 	zh:controller:endpoint: ZCL command 0x00124b000cb52894/3 genOnOff.defaultRsp({"cmdId":10,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":42,"writeUndiv":false})
[2024-06-03 19:31:06] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0x00124b000cb52894:30123/3 (0,0,1)
[2024-06-03 19:31:06] debug: 	zh:zstack:znp: SREQ: --> AF - dataRequest - {"dstaddr":30123,"destendpoint":3,"srcendpoint":1,"clusterid":6,"transid":175,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,42,11,10,0]}}
[2024-06-03 19:31:06] debug: 	zh:zstack:unpi:writer: --> frame [254,15,36,1,171,117,3,1,6,0,175,0,30,5,24,42,11,10,0,119]
[2024-06-03 19:31:06] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-06-03 19:31:06] debug: 	z2m: Received Zigbee message from '0x00124b000cb52894', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 3 with groupID 0
State changed {
  entity: Device {
    zh: Device {
      _events: [Object: null prototype] {},
      _eventsCount: 0,
      _maxListeners: undefined,
      ID: 37,
      _applicationVersion: 1,
      _dateCode: '2024-06-03',
      _endpoints: [Array],
      _hardwareVersion: 1,
      _ieeeAddr: '0x00124b000cb52894',
      _interviewCompleted: true,
      _interviewing: false,
      _lastSeen: 1717435866663,
      _manufacturerID: 64447,
      _manufacturerName: 'ptvo.info',
      _modelID: 'ptvotest',
      _networkAddress: 30123,
      _powerSource: 'Mains (single phase)',
      _softwareBuildID: '2024-06-03',
      _stackVersion: 1,
      _type: 'Router',
      _zclVersion: 1,
      _linkquality: 99,
      _skipDefaultResponse: false,
      _customReadResponse: undefined,
      _deleted: undefined,
      _lastDefaultResponseSequenceNumber: 42,
      _checkinInterval: undefined,
      _pendingRequestTimeout: 0,
      _customClusters: {},
      meta: [Object],
      [Symbol(kCapture)]: false
    },
    definition: {
      zigbeeModel: [Array],
      model: 'ptvotest',
      vendor: 'Custom devices (DiY)',
      description: '[Configurable firmware](https://ptvo.info/zigbee-configurable-firmware-features/)',
      fromZigbee: [Array],
      toZigbee: [Array],
      exposes: [Array],
      meta: [Object],
      endpoint: [Function: endpoint],
      configure: [AsyncFunction: configure],
      options: [Array]
    },
    _definitionModelID: 'ptvotest'
  },
  from: { state_l3: 'OFF', state_l1: 'OFF' },
  to: { state_l3: 'OFF', state_l1: 'OFF' },
  reason: null,
  update: { state_l3: 'OFF' }
}
[2024-06-03 19:31:06] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x00124b000cb52894', payload '{"current_l1":null,"linkquality":99,"power_l1":null,"state_l1":"OFF","state_l3":"OFF","voltage_l1":null}'
[2024-06-03 19:31:06] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
[2024-06-03 19:31:06] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
[2024-06-03 19:31:06] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
[2024-06-03 19:31:06] debug: 	zh:zstack:znp: SRSP: <-- AF - dataRequest - {"status":0}
[2024-06-03 19:31:06] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-06-03 19:31:06] debug: 	zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,175,105]
[2024-06-03 19:31:06] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,175,105]
[2024-06-03 19:31:06] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,175] - 105
[2024-06-03 19:31:06] debug: 	zh:zstack:znp: AREQ: <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":175}
[2024-06-03 19:31:06] debug: 	zh:zstack:unpi:parser: --- parseNext []

rickbhs, 2024-06-12:
@ptvoinfo can jou point me in the right direction?

ptvoinfo, 2024-06-23:
@rickbhs Sorry about the delay with my answer. 

I do not see any Zigbee data from INA216 in the logs. I think that the firmware cannot connect to your sensor and read data. Please check the I2C address, wires, pins definition in the firmware configuration.

closed 2024-08-19 by ptvoinfo