ptvoinfo/zigbee-configurable-firmware #286: trouble with external power sensor

alainvdu69, 2024-06-16:
Hi !
I've tested the external power sensor and it doesn't work as expected : it's allways on.
It goes down only when I flash a firmware on my E18-MS1-PCB.
I'm using the last version of PTVO, the 2024.05.23.

My config :
[Common]
Left=643
Top=180
Width=1106
Height=693
Board=0
Device type=3
LED active=True
LED inverted=False
Pin LED=P01
LED mode=3
Pin active 1=False
Pin No 1=P30
Pin Type 1=0
Pull Type 1=0
Pin inversed 1=False
State_1=False
Pin active 2=False
Pin No 2=P31
Pin Type 2=0
Pull Type 2=0
Pin inversed 2=False
State_2=True
Pin active 3=False
Pin No 3=P32
Pin Type 3=0
Pull Type 3=0
Pin inversed 3=False
State_3=True
Pin active 4=False
Pin No 4=P02
Pin Type 4=44
Pull Type 4=0
Pin inversed 4=False
State_4=True
Pin active 5=True
Pin No 5=P11
Pin Type 5=6
Pull Type 5=1
Pin inversed 5=False
State_5=True
Pin active 6=True
Pin No 6=P10
Pin Type 6=6
Pull Type 6=1
Pin inversed 6=False
State_6=True
Pin active 7=False
Pin No 7=P33
Pin Type 7=3
Pull Type 7=0
Pin inversed 7=False
State_7=True
Pin active 8=False
Pin No 8=P34
Pin Type 8=4
Pull Type 8=0
Pin inversed 8=False
State_8=True
Pin active 9=False
Pin Type 9=0
Pull Type 9=0
Pin inversed 9=False
State_9=True
Pin active 10=False
Pin Type 10=0
Pull Type 10=0
Pin inversed 10=False
State_10=True
Pin active 11=False
Pin Type 11=0
Pull Type 11=0
Pin inversed 11=False
State_11=True
Pin active 12=False
Pin Type 12=0
Pull Type 12=0
Pin inversed 12=False
State_12=True
Pin active 13=False
Pin Type 13=0
Pull Type 13=0
Pin inversed 13=False
State_13=True
Pin active 14=False
Pin Type 14=0
Pull Type 14=0
Pin inversed 14=False
State_14=True
Pin active 15=False
Pin Type 15=0
Pull Type 15=0
Pin inversed 15=False
State_15=True
Pin active 16=False
Pin Type 16=0
Pull Type 16=0
Pin inversed 16=False
State_16=True
Pin active 17=False
Pin No 17=P04
Pin Type 17=0
Pull Type 17=0
Pin inversed 17=False
Link_17=True
Switch_17=False
Hold_17=False
Double_17=False
Tripple_17=False
Bind type 17=3
Pin active 18=False
Pin No 18=P00
Pin Type 18=0
Pull Type 18=0
Pin inversed 18=False
Link_18=True
Switch_18=True
Hold_18=False
Double_18=False
Tripple_18=False
Bind type 18=0
Pin active 19=False
Pin No 19=P05
Pin Type 19=0
Pull Type 19=0
Pin inversed 19=False
Link_19=True
Switch_19=True
Hold_19=False
Double_19=False
Tripple_19=False
Bind type 19=0
Pin active 20=False
Pin No 20=P33
Pin Type 20=3
Pull Type 20=0
Pin inversed 20=False
Link_20=True
Switch_20=False
Hold_20=False
Double_20=False
Tripple_20=False
Bind type 20=0
Pin active 21=False
Pin No 21=P34
Pin Type 21=4
Pull Type 21=0
Pin inversed 21=False
Link_21=True
Switch_21=False
Hold_21=False
Double_21=False
Tripple_21=False
Bind type 21=0
Pin active 22=False
Pin Type 22=0
Pull Type 22=0
Pin inversed 22=False
Link_22=True
Switch_22=False
Hold_22=False
Double_22=False
Tripple_22=False
Bind type 22=0
Pin active 23=False
Pin Type 23=0
Pull Type 23=0
Pin inversed 23=False
Link_23=True
Switch_23=False
Hold_23=False
Double_23=False
Tripple_23=False
Bind type 23=0
Pin active 24=False
Pin Type 24=0
Pull Type 24=0
Pin inversed 24=False
Link_24=True
Switch_24=False
Hold_24=False
Double_24=False
Tripple_24=False
Bind type 24=0
Pin active 25=False
Pin Type 25=0
Pull Type 25=0
Pin inversed 25=False
Link_25=True
Switch_25=False
Hold_25=False
Double_25=False
Tripple_25=False
Bind type 25=0
Pin active 26=False
Pin Type 26=0
Pull Type 26=0
Pin inversed 26=False
Link_26=True
Switch_26=False
Hold_26=False
Double_26=False
Tripple_26=False
Bind type 26=0
Pin active 27=False
Pin Type 27=0
Pull Type 27=0
Pin inversed 27=False
Link_27=True
Switch_27=False
Hold_27=False
Double_27=False
Tripple_27=False
Bind type 27=0
Pin active 28=False
Pin Type 28=0
Pull Type 28=0
Pin inversed 28=False
Link_28=True
Switch_28=False
Hold_28=False
Double_28=False
Tripple_28=False
Bind type 28=0
Pin active 29=False
Pin Type 29=0
Pull Type 29=0
Pin inversed 29=False
Link_29=True
Switch_29=False
Hold_29=False
Double_29=False
Tripple_29=False
Bind type 29=0
Pin active 30=False
Pin Type 30=0
Pull Type 30=0
Pin inversed 30=False
Link_30=True
Switch_30=False
Hold_30=False
Double_30=False
Tripple_30=False
Bind type 30=0
Pin active 31=False
Pin Type 31=0
Pull Type 31=0
Pin inversed 31=False
Link_31=True
Switch_31=False
Hold_31=False
Double_31=False
Tripple_31=False
Bind type 31=0
Pin active 32=False
Pin Type 32=0
Pull Type 32=0
Pin inversed 32=False
Link_32=True
Switch_32=False
Hold_32=False
Double_32=False
Tripple_32=False
Bind type 32=0
Param7_1=9
Param7_2=1
Param7_3=1
Param1_4=1
Param2_4=0
Param1_5=2000
Param1_6=2000

[Expert]
ManufacturerIdEn=False
ManufacturerId=64447
ManufacturerNameEn=False
ManufacturerName=ptvo.info
ModelIdEn=True
ModelId=ptvo.switch
Icon enabled=False
LocationDescriptionEn=False
LinkKeyEn=False
LinkKey=01030507090B0D0F00020406080A0C0D
ReportIntEn=True
ReportInt=10
Disable report interval config=False
DateCode=True
WatchDog=False
PSM mode wait commands=True
PSM mode net poll enabled=False
PSM mode net poll interval=0
Disable reset=False

[PA]
TX EN=P11
RX EN=P10

[Expert.SBL]
Enabled=False

[Expert.OTA]

ptvoinfo, 2024-06-23:
@alainvdu69 Sorry about the delay with my answer.
I do not see any sensors defined in your configuration. The "External power control" function works only for sensors.

alainvdu69, 2024-06-24:
hi !
in fact, the sensor is a US100.
to check what's wrong when i detect that the power consumption where still near 2mA during the sleep mode, i've unchecked the us100

below, the real config, and with this one, a LED on P11 is always lighting ...

[Common]
Left=643
Top=180
Width=1106
Height=693
Board=0
Device type=3
LED active=True
LED inverted=False
Pin LED=P01
LED mode=3
Pin active 1=False
Pin No 1=P30
Pin Type 1=0
Pull Type 1=0
Pin inversed 1=False
State_1=False
Pin active 2=False
Pin No 2=P31
Pin Type 2=0
Pull Type 2=0
Pin inversed 2=False
State_2=True
Pin active 3=False
Pin No 3=P32
Pin Type 3=0
Pull Type 3=0
Pin inversed 3=False
State_3=True
Pin active 4=True
Pin No 4=P02
Pin Type 4=44
Pull Type 4=0
Pin inversed 4=False
State_4=True
Pin active 5=True
Pin No 5=P11
Pin Type 5=6
Pull Type 5=1
Pin inversed 5=False
State_5=True
Pin active 6=False
Pin No 6=P10
Pin Type 6=6
Pull Type 6=1
Pin inversed 6=False
State_6=True
Pin active 7=False
Pin No 7=P33
Pin Type 7=3
Pull Type 7=0
Pin inversed 7=False
State_7=True
Pin active 8=False
Pin No 8=P34
Pin Type 8=4
Pull Type 8=0
Pin inversed 8=False
State_8=True
Pin active 9=False
Pin Type 9=0
Pull Type 9=0
Pin inversed 9=False
State_9=True
Pin active 10=False
Pin Type 10=0
Pull Type 10=0
Pin inversed 10=False
State_10=True
Pin active 11=False
Pin Type 11=0
Pull Type 11=0
Pin inversed 11=False
State_11=True
Pin active 12=False
Pin Type 12=0
Pull Type 12=0
Pin inversed 12=False
State_12=True
Pin active 13=False
Pin Type 13=0
Pull Type 13=0
Pin inversed 13=False
State_13=True
Pin active 14=False
Pin Type 14=0
Pull Type 14=0
Pin inversed 14=False
State_14=True
Pin active 15=False
Pin Type 15=0
Pull Type 15=0
Pin inversed 15=False
State_15=True
Pin active 16=False
Pin Type 16=0
Pull Type 16=0
Pin inversed 16=False
State_16=True
Pin active 17=False
Pin No 17=P04
Pin Type 17=0
Pull Type 17=0
Pin inversed 17=False
Link_17=True
Switch_17=False
Hold_17=False
Double_17=False
Tripple_17=False
Bind type 17=3
Pin active 18=False
Pin No 18=P00
Pin Type 18=0
Pull Type 18=0
Pin inversed 18=False
Link_18=True
Switch_18=True
Hold_18=False
Double_18=False
Tripple_18=False
Bind type 18=0
Pin active 19=False
Pin No 19=P05
Pin Type 19=0
Pull Type 19=0
Pin inversed 19=False
Link_19=True
Switch_19=True
Hold_19=False
Double_19=False
Tripple_19=False
Bind type 19=0
Pin active 20=False
Pin No 20=P33
Pin Type 20=3
Pull Type 20=0
Pin inversed 20=False
Link_20=True
Switch_20=False
Hold_20=False
Double_20=False
Tripple_20=False
Bind type 20=0
Pin active 21=False
Pin No 21=P34
Pin Type 21=4
Pull Type 21=0
Pin inversed 21=False
Link_21=True
Switch_21=False
Hold_21=False
Double_21=False
Tripple_21=False
Bind type 21=0
Pin active 22=False
Pin Type 22=0
Pull Type 22=0
Pin inversed 22=False
Link_22=True
Switch_22=False
Hold_22=False
Double_22=False
Tripple_22=False
Bind type 22=0
Pin active 23=False
Pin Type 23=0
Pull Type 23=0
Pin inversed 23=False
Link_23=True
Switch_23=False
Hold_23=False
Double_23=False
Tripple_23=False
Bind type 23=0
Pin active 24=False
Pin Type 24=0
Pull Type 24=0
Pin inversed 24=False
Link_24=True
Switch_24=False
Hold_24=False
Double_24=False
Tripple_24=False
Bind type 24=0
Pin active 25=False
Pin Type 25=0
Pull Type 25=0
Pin inversed 25=False
Link_25=True
Switch_25=False
Hold_25=False
Double_25=False
Tripple_25=False
Bind type 25=0
Pin active 26=False
Pin Type 26=0
Pull Type 26=0
Pin inversed 26=False
Link_26=True
Switch_26=False
Hold_26=False
Double_26=False
Tripple_26=False
Bind type 26=0
Pin active 27=False
Pin Type 27=0
Pull Type 27=0
Pin inversed 27=False
Link_27=True
Switch_27=False
Hold_27=False
Double_27=False
Tripple_27=False
Bind type 27=0
Pin active 28=False
Pin Type 28=0
Pull Type 28=0
Pin inversed 28=False
Link_28=True
Switch_28=False
Hold_28=False
Double_28=False
Tripple_28=False
Bind type 28=0
Pin active 29=False
Pin Type 29=0
Pull Type 29=0
Pin inversed 29=False
Link_29=True
Switch_29=False
Hold_29=False
Double_29=False
Tripple_29=False
Bind type 29=0
Pin active 30=False
Pin Type 30=0
Pull Type 30=0
Pin inversed 30=False
Link_30=True
Switch_30=False
Hold_30=False
Double_30=False
Tripple_30=False
Bind type 30=0
Pin active 31=False
Pin Type 31=0
Pull Type 31=0
Pin inversed 31=False
Link_31=True
Switch_31=False
Hold_31=False
Double_31=False
Tripple_31=False
Bind type 31=0
Pin active 32=False
Pin Type 32=0
Pull Type 32=0
Pin inversed 32=False
Link_32=True
Switch_32=False
Hold_32=False
Double_32=False
Tripple_32=False
Bind type 32=0
Param7_1=9
Param7_2=1
Param7_3=1
Param1_4=1
Param2_4=0
Param1_5=2000
Param1_6=2000

[Expert]
ManufacturerIdEn=False
ManufacturerId=64447
ManufacturerNameEn=False
ManufacturerName=ptvo.info
ModelIdEn=True
ModelId=ptvo.switch
Icon enabled=False
LocationDescriptionEn=False
LinkKeyEn=False
LinkKey=01030507090B0D0F00020406080A0C0D
ReportIntEn=True
ReportInt=10
Disable report interval config=False
DateCode=True
WatchDog=False
PSM mode wait commands=True
PSM mode net poll enabled=False
PSM mode net poll interval=0
Disable reset=False

[PA]
TX EN=P11
RX EN=P10

[Expert.SBL]
Enabled=False

[Expert.OTA]

ptvoinfo, 2024-07-13:
The external sensor power control sensor should be immediately before a controllable sensor.
[image]

alainvdu69, 2024-07-14:
yes, it's working now.
thanks.

closed 2024-07-14 by alainvdu69