ptvoinfo/zigbee-configurable-firmware #295: HC-SR04T V3

BMWK1, 2024-07-08:
Hi, and thank you for your work

Configuring an HC-SR04T like the exemple.
Led flashes every 60s, MQTT explorer returns : {"l1":0,"linkquality":236}, so no available data

I have seen on the website this:
Hi, there is a version of jsn-sr04t-v3-0, that too has a serial connection, but a little different:
Serial mode operates at 9600 baud and works similar to serial mode on the US-100 module. The main difference is that it also returns a start byte and checksum byte. It also can be operated in an automatic mode of operation where the module is automatically triggered every 100mS.
The way Serial mode works is as follows:

The module is normally idle.
A command bye of 0x55 on the RX pin initiates a detection cycle.
The module responds with 4 bytes of information
Start bye of 0xFF
Most Significant Byte (MSB)
Least Significant Byte (LSB)
Checksum Byte

The returned value can be calculated using the formula (MSB * 256) + LSB.
The checksum byte is calculated using 0xFF + MSB + LSB. It returns the least significant byte of the calculation only.
It seems, that ptvo uses just the first two bytes, that is 255 and MSB and does not read the rest.

ptvoinfo, 2024-07-08:
@BMWK1 So, you should find the correct command and response type for your device. Or test other variant in the firmware settings.

BMWK1, 2024-07-11:
I have finally successfully set up my JSN-SR4T V3, after i had downloaded the very last version of ptvoinfo....
For this sensor, the right setting is using ultrasonic UART sensor, JSNO4T, not JSN04T V2 :
[image]

as the command is 0x55, and the return sequence is  4 bytes of information
1. Start with 0xFF
2. Most Significant Byte (MSB)
3. Least Significant Byte (LSB)
4. Checksum Byte
The returned value can be calculated using the formula (MSB * 256) + LSB.
The checksum byte is calculated using 0xFF + MSB + LSB. It returns the least significant byte of
the calculation only.

closed 2024-07-11 by BMWK1