ptvoinfo/zigbee-configurable-firmware #296: Value glitch double i2c

TomasWeij, 2024-07-09:
Hi,

I've added two HX711 sensors to my firmware config. See screenshot below. My report interval is every 2s. One input shows the value and updates it every 2s as expected. The other value jumps between 0 and the actual value. The firmware has been flashed on the RF-Star CC2652P2.
[bedmat_no_mmwave.txt](https://github.com/user-attachments/files/16149917/bedmat_no_mmwave.txt)
[image]

TomasWeij, 2024-07-12:
I'm willing to compensate you, please contact me.

ptvoinfo, 2024-07-18:
@TomasWeij I've tested this configuration. Please try to shift P21 and P13 down (to Output 4 & Output 5). Leave one unused Output 3 between sensors in the configuration. After this, you'll need to re-create a custom converter (if used).

closed 2024-09-15 by ptvoinfo