ptvoinfo/zigbee-configurable-firmware #299: PZEM-004 V3 + CC2530

bdariusb, 2024-07-17:
I have an issue with reported power results. I used the latest firmware, z2m 1.39.0 and generated z2m custom converter. If you look in the attached picture, the voltage and current, the calculated power is much higher than expected (~10x higher?)
[image]
[image]

ptvoinfo, 2024-07-19:
@bdariusb Could you please download and try the latest version? You should re-flash the firmware only.

ptvoinfo, 2024-07-26:
@bdariusb I've removed this parameter from reporting because it is a constant 99.99% of time.

bdariusb, 2024-07-26:
> @bdariusb I've removed this parameter from reporting because it is a constant 99.99% of time.

Understandable, thanks

closed 2024-07-26 by bdariusb