ptvoinfo/zigbee-configurable-firmware #301: Cannot join CC2530 with PSM mode

ywwzwb, 2024-08-03:
chip: CC2530F256
Zigbee2MQTT version 1.39.1
Coordinator type: zStack3x0
Coordinator revision: 20230507
I got some device with CC2350 chips,  just 7 keys on board, work as a remote controller, power by batteries. 
I've  tried with plain end device and it works just fine, and I paied a license to using PSM mode to save batteries. 
The PSM config is same as the end device except device type is PSM, after flash the PSM firmware and the device can not join the network, the zigbee2mqtt keep saying "Interview failed because can not get active endpoints".
Here is my config.(I have tried the "CC2530 without external oscillator and just same issue")
[image]
the error is here.
[image]
[image]
the image of pcb
[image]

[image]

the logs 
```
info 2024-08-03 23:21:01zh:controller: Interview for '0x00124b000ff0edf6' started
info 2024-08-03 23:21:01z2m: Starting interview of '0x00124b000ff0edf6'
info 2024-08-03 23:21:01z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b000ff0edf6","ieee_address":"0x00124b000ff0edf6","status":"started"},"type":"device_interview"}'
error 2024-08-03 23:21:21zh:controller: Interview failed for '0x00124b000ff0edf6 with error 'Error: Interview failed because can not get active endpoints ('0x00124b000ff0edf6')'
error 2024-08-03 23:21:21z2m: Failed to interview '0x00124b000ff0edf6', device has not successfully been paired
info 2024-08-03 23:21:21z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00124b000ff0edf6","ieee_address":"0x00124b000ff0edf6","status":"failed"},"type":"device_interview"}'
```

MB901, 2024-08-13:
Hi, 
I have exactly same probleme.
CC2530 only in PSM mode too. 
Did you found a solution ?

ptvoinfo, 2024-08-14:
@ywwzwb Did you try to press any button periodically while joining? 
Try to decrease the number of buttons in the configuration.

ywwzwb, 2024-08-16:
@ptvoinfo I tried to decrease the number of buttons to two last night，and it works. It was weird and  I want find how many buttons are support at most, so I increase the number of buttons, finally, all buttons were in the configuration, and it still working. I even tried the firmware when I opened this issue, and it still works. It was weird and I has no idea, one thing may have some helps, when I was flashing the firmware, the data cable pulled the computer by a few centimeters, which may have moved both the coordinator and its antenna. 
Well, It's working for now, and I tried some other device and all device work fine.

ptvoinfo, 2024-08-17:
@ywwzwb I'm sure, 4 inputs work without any problem, but it depends on a schematic.

Are you sure that all inputs exist and have the external pull-up resistor?

Also, try to disable the status led indicator, or input with P04 because the LED indicator uses the internal pull-up resistor, it is enabling the internal pull-up resistor for P04.

ywwzwb, 2024-08-20:
@ptvoinfo Yes， I'm quite sure that all input buttons have external pull-up resistor, here is the diagram for that.
[image]
It's working  for now, and I can't repeat this issue again. I've checked every thing, but nothing really changed. Becase I ran the zig2mqtt and Windows on one vmware system, and the coordinator using usb port, so the the device is not far from the coordinator while flashing, may be 1 meter at most.

closed 2024-08-26 by ptvoinfo