ptvoinfo/zigbee-configurable-firmware #303: Counter sensor doesn't work for SNZB-04's reed switch

xrucka, 2024-08-22:
Hello, 

I'm currently using SNZB-04 dry contact sensor to report magnetic pulses to Z2M and HASS. However my sensor is at far end of the mesh and sometimes the messages get lost. I was hoping to use the counter sensor to store the pulse count on the SNZB-04 and report total count intead of individual pulses, however I'm not able to increment the counter. However, if I use GPIO instead of counter sensor, the pulses get reported sufficiently (thus reproducing the functionality of the stock firmware).

Button is connected to P20, reed switch to P06. Both have an external pullup.
* If I setup either as GPIO, the pulse/push is reported.
* If I setup P20 as counter (input section), counter is incremented with each press.
* If I setup P06 as counter (input section), counter is never incremented.
* If I setup either as GPIO and link it to coresponding output (configured as counter with parameters), button/reed triggers and yet, no increment in counter occurs.

I had tested 6 versions of the firmware, starting with 7674309 up to the current 536b91a, yet all behave the same. 

What is the correct way to setup a counter for such a scenario?

ptvoinfo, 2024-08-26:
Please try to configure a single counter as shown here:
https://ptvo.info/zigbee-configurable-firmware-features/pulse-generator-and-counter/

xrucka, 2024-08-26:
@ptvoinfo Hello and thank you for your reply. Just to clarify, the page you refer to (as far as I'm aware) defines two counters - one for P12 and one for P13, plus pulse generator on P00 - is this interpretation correct? Is the pulse generator (in any way) related to pulse counter, as whether the counter can count only internal or only external pulses?

ptvoinfo, 2024-08-26:
No, the pulse counter and pulse generator are independent. In your case, you can configure just one counter with the desired settings.

xrucka, 2024-08-26:
Ok, I managed to set up the device as requested, first using P20 (button, 10ms debounce) and further P06 (reed switch, all debounce intervals, exactly the same as previous experiment except mentioned) and P06 as GPIO input 1 (to confirm that the reed switch is sufficiently activated in given test setup assembly).

[image]

First experiment (button counter) worked as expected, except for potential double increment (I expect fine-tuning debounce would solve this). 

I was not able to find debounce configuration that would make the reed switch counter mode work.

I was able to detect magnet presence using P06 as GPIO switch.

---

Is there any expectation about shape/stability of the pulse, other than debounce configuration?

ptvoinfo, 2024-09-05:
@xrucka Sorry, but you need a hardware pulse debounce solution.

xrucka, 2024-09-05:
@ptvoinfo Ok, I'll try to tweak the module a bit (there are some exposed pads I could possibly tap into). Furthermore, would it eventualy be possible to set virtual pin as output and use the same virtual pin as input (thus, creating a loopback channel)?

ptvoinfo, 2024-09-05:
@xrucka It is impossible too.

closed 2024-09-28 by ptvoinfo