ptvoinfo/zigbee-configurable-firmware #304: Better explanation needed for the use of input 1 for a pairing button

rmtucker, 2024-08-29:
Still waiting on my CC2530 and programmer arriving but having trouble with firmware config:
I wish to have 2 DS18B20 sensors,1 on output 1,and 1 on output 2 as pic
[image]
But then if i try to add input 1 external wake up it gives the following error
[image]

ptvoinfo, 2024-09-05:
@rmtucker You cannot combine DS18B20 and "External wakeup" on the same line #1 (endpoint #1). It is the internal limitation.
You can shift DS18B20 to Output 2 & 3.

rmtucker, 2024-09-05:
I understand thank you
But is external wakeup the correct choice for a pairing button,or should it be just as switch on gpio?

ptvoinfo, 2024-09-08:
The External wakeup button on Input 1 can also work as a paring button.

closed 2024-09-28 by ptvoinfo