ptvoinfo/zigbee-configurable-firmware #305: Icon is not displayed in Z2M

Maschler, 2024-09-08:
Hi!
First of all, congratulations on the great software and thanks for sharing!!! I was able to implement what I wanted to do with it and I'm really excited.

However, I noticed two things. To keep a better overview, I have opened two issues.
(Zigbee2MQTT version: 1.39.1, PTVO version: 2.5)

1. I would like to display an icon in Z2M. I have downloaded the desired .png (see fig 1) and loaded it under “Expert” (see fig 2).

2. Then I placed the generated Test_26.js for Z2M in the DATA folder and coupled the flashed CC2530 (which worked great)
3. Unfortunately no icon is displayed in Z2M (see fig 3).

Was my procedure wrong? Is the .png too big (are there any limits)?

**Test_26.js**
[Test_26js.txt](https://github.com/user-attachments/files/16921538/Test_26js.txt)
**fig 1**
[image]
**fig 2**
[image]
**fig 3**
[image]

ptvoinfo, 2024-09-08:
@Maschler Did you add your converter to the Z2M configuration? Did you try to restart Z2M afte it?

Maschler, 2024-09-09:
Thank you for your answer.

I used the Costom Converter like in Fig 2 shown. Afterwards I then transferred the generated Test_26.js to the Z2M data folder and restarted Z2M. After this I paired the cc2530.

Is the size of the picture to large? (Dimensions 3000x1875 pixels)

ptvoinfo, 2024-09-09:
It is not enough. You should add your converter to the Z2M configuration.
configuration.yaml -> external_converters:

Maschler, 2024-09-09:
Ohhh. This solved both issues. Once again, the problem was sitting in front of the screen... Thank you very much!

closed 2024-09-09 by Maschler