ptvoinfo/zigbee-configurable-firmware #306: Contact sensor is displayed as a switch

Maschler, 2024-09-08:
Zigbee2MQTT Version: 1.39.1, PTVO Version: 2.5
Here is my second problem:

1. I have configured an input as a switch. If 3.3V is applied to it, this should be reported (--> status: Open or Closed).
2. To do this, I connected the input to a virtual output and declared the output as “Contact” (see **Test_26.txt**).
3. I expected that Z2M would now display Status: Open or Closed like a window contact sensor (example: Aqara MCCGQ11LM **Fig 1**)
4. However, I see a switch that changes to On or Off depending on the voltage (see **Fig 2**).

Is my configuration wrong?
**Test_26.txt**
[Test_26.txt](https://github.com/user-attachments/files/16921553/Test_26.txt)
**Fig 1**
[image]
**Fig **
[image]

ptvoinfo, 2024-09-08:
Please try to create a custom converter for your device and add it to Z2M.

Maschler, 2024-09-09:
Thank you for your answer.

I did this via your software using the Costom Converter ([here](https://private-user-images.githubusercontent.com/38862834/365424408-45bf843f-5420-4ab6-b21c-421e529d0e27.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU4NjYxOTUsIm5iZiI6MTcyNTg2NTg5NSwicGF0aCI6Ii8zODg2MjgzNC8zNjU0MjQ0MDgtNDViZjg0M2YtNTQyMC00YWI2LWIyMWMtNDIxZTUyOWQwZTI3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MDklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTA5VDA3MTEzNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWI0YTMxOWMyMTdmYTc2NWNmYzE4OGVkODE5ZDM1MTEwNmI3MDI3YjI4ODc4NThmMWUzMjcyNmFjMWEzZWQwNTAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.uA4agpg1h6G49BCk0TjroI4VKMLQsn11unIoAt2dXIw)).

I then transferred it to the Z2M data folder and restarted Z2M. Then I did the paring of the cc2530. The resulte is visible in Fig 2.

closed 2024-09-09 by Maschler