ptvoinfo/zigbee-configurable-firmware #307: Dynamic sensor support in HA

daestx, 2024-09-09:
I'd like to use a CC2530 device in combination with the PVTO firmware as a generic UART interface to transmit data from a microcontroller to Home Assistant.

My first test is running successfully. I have configured a firmware that supports P02 (connected to TX from the microcontroller) and P03 (connected to RX from the MC). I have also generated the connector script with the tool.

When I send data, it is displayed in the “Action” control element.

[image]

However, my use case is that I want to transfer any data from the MC without having to adjust the configuration of the connector every time.

I could imagine transmitting a string via the UART that describes which data is involved. In a simple case this could be, for example, “Sensor1,5,rpm”, “Sensor2,5,C”. The corresponding sensor should then be automatically created in the MQTT device and the measured values transferred

Is it possible to include a parser in the connector script that assigns sensor data accordingly and dynamically creates sensors if they do not yet exist?

ptvoinfo, 2024-09-09:
@daestx This question is more related to Zigbee2Mqtt than my firmware. 
The parser cannot create sensors in run-time. They should be defined explicitly. Otherwise, Z2M cannot send info about sensors to HA.

daestx, 2024-09-09:
Yeah, I wasn't sure if this was the correct forum for my question.

At the moment the structure of the converter scripts is not clear to me either. Even if I were to specify a fixed sensor configuration (via exposes???) I don't know where to put the parser to perform the distribution of the data to the sensor elements

ptvoinfo, 2024-09-09:
You should override (redefine) the "ptvo_switch_uart" (defined in zigbee-herdsman-converters => converters => fromZigbee.js) in your code.

daestx, 2024-09-09:
Yes, I ended up at the same point. Currently it is defined like this:

ptvo_switch_uart: {
    cluster: 'genMultistateValue',
    type: ['attributeReport', 'readResponse'],
    convert: (model, msg, publish, options, meta) => {
        let data = msg.data['stateText'];
        if (typeof data === 'object') {
            let bHex = false;
            let code;
            let index;
            for (index = 0; index < data.length; index += 1) {
                code = data[index];
                if (code < 32 || code > 127) {
                    bHex = true;
                    break;
                }
            }
            if (!bHex) {
                data = data.toString('latin1');
            } else {
                data = [...data];
            }
        }
        return {action: data};
    },
} 

It seems like all data are copied to "action". This function is a part of a type script file. Do you have an example of a override function for a converter written in js?

daestx, 2024-09-09:
Ok, I got it. I just added

fz.ptvo_switch_uart = {
    cluster: 'genMultistateValue',
    type: ['attributeReport', 'readResponse'],
    convert: (model, msg, publish, options, meta) => {
        let data = msg.data['stateText'];
        if (typeof data === 'object') {
            let bHex = false;
            let code;
            let index;
            for (index = 0; index < data.length; index += 1) {
                code = data[index];
                if (code < 32 || code > 127) {
                    bHex = true;
                    break;
                }
            }
            if (!bHex) {
                data = data.toString('latin1');
            } else {
                data = [...data];
            }
        }
        return {action: "this is a test", wind_strength: 1};
    },
} 

and now, I see that my sensor: exposes.numeric('wind_strength', ea.STATE_SET).withDescription('some sensor'),
switched to 1.

closed 2024-09-09 by daestx