ptvoinfo/zigbee-configurable-firmware #311: z2m loads external converter but doesnt seem to apply it?

gingemonster, 2024-12-02:
I have specified the custom convert .js file (created in the firmware app) in the z2m settings and when I reload z2m I get:

[2024-12-02 13:44:39] info: 	z2m: Loaded external converter 'GM.KeyHook1.js'

the converter js has this line to match by model:

const ptvo_switch = (zigbeeHerdsmanConverters.findByModel)?zigbeeHerdsmanConverters.findByModel('ptvo.switch'):zigbeeHerdsmanConverters.findByDevice({modelID: 'ptvo.switch'});

and z2m shows that as my devices model:

[image]

Yet the custom description, manufacturer etc I set in the js is not getting applied:

const device = {
    zigbeeModel: ['GM.KeyHook1'],
    model: 'GM.KeyHook1',
    vendor: 'Gingemonster',
    description: 'TEST [Configurable firmware](https://ptvo.info/zigbee-configurable-firmware-features/)',
    
  Any ideas?

gingemonster, 2024-12-03:
Ok found the issue, although I have these settings in the advanced of the firmware:

[image]

My device is still reporting itself as a ptvo.switch so the converter js exported from the firmware tool does not match the device. Manually reverting it back to the defaults made the converter load correctly.

Now things are working as expected except there are a lot of additional controls, settings and sensors exposed to home assistant that Im sure are correct I just dont really understand what they represent:

[image]

A future article on the documentation expanding a bit more on what these are and how to hide them if not needed would be much appreciated

Thanjs

alainvdu69, 2024-12-05:
Hi !
I've big trouble with external converter too.
At this time, i'm not using it ... 
But, perhaps it's the same trouble with the powersource issue you have.
So, try to delete, restart z2m, and repair with the fresh firmware with the converter enabled.

I will make further experiments to find the right solution.

gingemonster, 2024-12-08:
yes was z2m caching previous entry when on batteries and not the latest flashed firmware

closed 2024-12-08 by gingemonster