ptvoinfo/zigbee-configurable-firmware #313: messages repeated six times

alainvdu69, 2024-12-05:
Hi !

I've noticed that all my ptvo devices (3 with PZEM-004T on AC/DC power) (3 in PSM mode on battery) repeat the zigbee message 6 six times.
I've discovered this with MQTT Explorer.
For exemple, on a PSM device, with à periodic report of 30 minutes :
we can see a flow of 6 messages with a very short timelaps, and 30 minutes before, the flow of the past 6 messages.

[image]
[image]
[image]
[image]

It's very strange.
my other devices (SONOFF, NOUS, ORVIBO, ...) don't have the repetition of the message.

the firmware for the PSM devices was build with the PTVO from july 2024 version, and the others devices have a more older version of PTVO (year 2022.)

ptvoinfo, 2024-12-06:
The device sends one Zigbee data packet per a value. So, you see these separate messages in Z2M and MQTT. Set the "debounce" parameter to "1" in the device settings in Z2M. Then Z2M will group these messages.

alainvdu69, 2024-12-06:
yes, well done.
solved.
thanks.

closed 2024-12-06 by alainvdu69

alainvdu69, 2024-12-06:
but, there were a drawback ...
sometimes, the device send usefull messages in a very short laptime, and the debounce to 1 drop them.
so, i have to add the relevant id of values to the 'ignore anti-rebond' :
[image]