ptvoinfo/zigbee-configurable-firmware #314: Pulse switch non returning to Off

frabnet, 2024-12-07:
Hi, I'm trying to configure two outputs as pulse switches.

When I turn ON the Output 5, I see the respective GPIO turning ON then returning to OFF, so hardware side it works as expected.

But in Zigbee2Mqtt dashboard the state stays ON even if the GPIO already went back to OFF.

```
    "state_l5": "ON",
    "state_l6": "OFF",

```

Did I miss something about the  "Change to sate after interval" part or did I just found a bug?

Thank you.

[image]

ptvoinfo, 2024-12-08:
@frabnet Hi!

Please try the latest version of the firmware released today.

frabnet, 2024-12-08:
Ok, I completely missed those values.

The window did not resize to let me see, but maybe that's on me, I use FirmwareConfig.exe with Ubuntu through Wine emulation system.

[image]

Setting Interval fixed the "problem" (even with ptvo-firmware-2024-10-27), but a little bug remains.

The status returns to OFF in the mqtt message, in Status tab in Zigbee2Mqtt, in Home Assistant too, but in the Values tab of Zigbee2Mqtt it stays "ON" even if click the "update" arrows, see screenshot below.

Maybe it's a problem with Zigbee2Mqtt software?

I think we can close this bug here. Sorry.

```
    "state_l5": "OFF",
    "state_l6": "OFF",

```
[image]

ptvoinfo, 2024-12-09:
@frabnet 
> Maybe it's a problem with Zigbee2Mqtt software?

Yes, it is a known issue of Z2M.

closed 2025-01-13 by frabnet