ptvoinfo/zigbee-configurable-firmware #315: Home Assistant not accepting PTVO quirk

matsjrosenberg, 2024-12-08:
Hello,
I am experiencing a constant problem that Home Assistant does no longer recognise the ZHA quirks generated by PTVO – not even for flashes generated and working in the past. No changes to .py files or CC2530 config made (i.e. mirror 1:1), and I have tried all kinds of HA restarts/delete old entities combinations between each step, and after ZHA device configuration. Still shows up without quirk config. Any idea where I should I continue trouble shooting? I run the latest version of HA. 

Many thanks in advance, Mats

ptvoinfo, 2024-12-08:
@matsjrosenberg, Hi! Please attach a screenshot with your most simple configuration that you've tried to add to HA.

matsjrosenberg, 2024-12-08:
[image]
[image]

Hi! Thanks for responding! This config for example.  As mentioned I have also tried with configs that I know have been working in the past on the same HA system (although earlier versions) and the .py have not been recognised. The settings of the .py directory are correct and working for other quirks.

ptvoinfo, 2024-12-08:
@matsjrosenberg Please test the update:
https://ptvo.info/download/ptvo-firmware-2024-12-08.zip?v2

matsjrosenberg, 2024-12-08:
Thank you very much for this update - the .py is recognised now. And thank you for the very quick fix. Will buy some more licences now!!!

closed 2024-12-08 by matsjrosenberg