ptvoinfo/zigbee-configurable-firmware #318: Zighbee2mqtt TypeError: Cannot read properties of undefined (reading 'options')

irakhlin, 2025-01-07:
Using zigbee2mqtt 2.0.2, custom zigbee temperature device, using thew newest PTVO firmware configuration tool (1.9.0.28 Firmware 2024-12-08, zip: ptvo-firmware-2024-12-14.zip) I created a new zigbee2mqtt converter. This same converter works just fine under zigbee2mqtt 1.4.2.

Error thrown during start up:
[2025-01-07 11:57:59] error: 	z2m: Failed to call 'ExternalConverters' 'start' (TypeError: Cannot read properties of undefined (reading 'options')
    at prepareDefinition (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/index.ts:356:23)
    at addDefinition (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/index.ts:415:18)
    at ExternalConverters.loadJS (/app/lib/extension/externalConverters.ts:49:30)
    at ExternalConverters.loadFiles (/app/lib/extension/externalJS.ts:156:24)
    at ExternalConverters.start (/app/lib/extension/externalJS.ts:45:20)
    at processTicksAndRejections (node:internal/process/task_queues:105:5)
    at Controller.callExtensions (/app/lib/controller.ts:368:17)
    at Controller.start (/app/lib/controller.ts:186:9)
    at start (/app/index.js:161:5))

I am attaching the converter here, only changed/removed strings are for zigbeeModel, model and vendor.

```js
const zigbeeHerdsmanConverters = require('zigbee-herdsman-converters');
const zigbeeHerdsmanUtils = require('zigbee-herdsman-converters/lib/utils');

const exposes = zigbeeHerdsmanConverters['exposes'] || require("zigbee-herdsman-converters/lib/exposes");
const ea = exposes.access;
const e = exposes.presets;
const fz = zigbeeHerdsmanConverters.fromZigbeeConverters || zigbeeHerdsmanConverters.fromZigbee;
const tz = zigbeeHerdsmanConverters.toZigbeeConverters || zigbeeHerdsmanConverters.toZigbee;

const ptvo_switch = (zigbeeHerdsmanConverters.findByModel)?zigbeeHerdsmanConverters.findByModel('ptvo.switch'):zigbeeHerdsmanConverters.findByDevice({modelID: 'ptvo.switch'});
fz.legacy = ptvo_switch.meta.tuyaThermostatPreset;
fz.ptvo_on_off = {
  cluster: 'genOnOff',
  type: ['attributeReport', 'readResponse'],
  convert: (model, msg, publish, options, meta) => {
      if (msg.data.hasOwnProperty('onOff')) {
          const channel = msg.endpoint.ID;
          const endpointName = `l${channel}`;
          const binaryEndpoint = model.meta && model.meta.binaryEndpoints && model.meta.binaryEndpoints[endpointName];
          const prefix = (binaryEndpoint) ? model.meta.binaryEndpoints[endpointName] : 'state';
          const property = `${prefix}_${endpointName}`;
	  if (binaryEndpoint) {
            return {[property]: msg.data['onOff'] === 1};
          }
          return {[property]: msg.data['onOff'] === 1 ? 'ON' : 'OFF'};
      }
  },
};

// Copied from fromZigbee.js
// The standard converter does not append the endpoint number

fz.ptvo_humidity = {
  cluster: 'msRelativeHumidity',
  type: ['attributeReport', 'readResponse'],
  options: [exposes.options.precision('humidity'), exposes.options.calibration('humidity')],
  convert: (model, msg, publish, options, meta) => {
      const humidity = parseFloat(msg.data['measuredValue']) / 100.0;

      // https://github.com/Koenkk/zigbee2mqtt/issues/798
      // Sometimes the sensor publishes non-realistic vales, it should only publish message
      // in the 0 - 100 range, don't produce messages beyond these values.
      if (humidity >= 0 && humidity <= 100) {
          const property = zigbeeHerdsmanUtils.postfixWithEndpointName('humidity', msg, model, meta);
          return {[property]: zigbeeHerdsmanUtils.calibrateAndPrecisionRoundOptions(humidity, options, 'humidity')};
      }
  },
};

fz.ptvo_pressure = {
  cluster: 'msPressureMeasurement',
  type: ['attributeReport', 'readResponse'],
  options: [exposes.options.precision('pressure'), exposes.options.calibration('pressure')],
[... 146 lines cut ...]
    exposes: [exposes.numeric('l3', ea.STATE_SET).withDescription('Signal strength level (0..20)'),
      e.temperature().withEndpoint('l4'),
      e.humidity().withEndpoint('l4'),
      e.switch().withEndpoint('l6'),
      e.action(['single', 'double', 'triple', 'hold', 'release']),
      ...ptvo_on_off_config_exposes('l6'),
],
    meta: {
        multiEndpoint: true,
        
    },
    endpoint: (device) => {
        return {
            l3: 3, l4: 4, l6: 6,
        };
    },
    configure: async (device, coordinatorEndpoint, logger) => {
      const endpoint = device.getEndpoint(1);
      await endpoint.read('genBasic', ['modelId', 'swBuildId', 'powerSource']);
    },

};

module.exports = device;
```

ptvoinfo, 2025-01-09:
@irakhlin Hi! Are you sure that this error message in the log refers to this converter?

irakhlin, 2025-01-11:
@ptvoinfo 
I am fairly certain, I just reupdated my zigbee2mqtt to version 2 and removed all converters I had in the external_converters directory. The following is the error during start up:
```
[2025-01-11 00:40:50] error: 	z2m: Failed to load external converter 'climatesense2.js'
[2025-01-11 00:40:50] error: 	z2m: Check the code for syntax error and make sure it is up to date with the current Zigbee2MQTT version.
[2025-01-11 00:40:50] error: 	z2m: External converters are not meant for long term usage, but for local testing after which a pull request should be created to add out-of-the-box support for the device
[2025-01-11 00:40:50] error: 	z2m: Failed to call 'ExternalConverters' 'start' (TypeError: Cannot read properties of undefined (reading 'options')
    at prepareDefinition (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/index.ts:356:23)
    at addDefinition (/app/node_modules/.pnpm/zigbee-herdsman-converters@21.12.0/node_modules/src/index.ts:415:18)
    at ExternalConverters.loadJS (/app/lib/extension/externalConverters.ts:49:30)
    at ExternalConverters.loadFiles (/app/lib/extension/externalJS.ts:156:24)
    at ExternalConverters.start (/app/lib/extension/externalJS.ts:45:20)
    at processTicksAndRejections (node:internal/process/task_queues:105:5)
    at Controller.callExtensions (/app/lib/controller.ts:368:17)
    at Controller.start (/app/lib/controller.ts:186:9)
    at start (/app/index.js:161:5))
```

All devices using that converter show up as Unsupported.

ptvoinfo, 2025-01-11:
@irakhlin I've tested your converter in the latest Z2M dev version, and Z2M can load it successfully without any modifications.

irakhlin, 2025-01-11:
@ptvoinfo  just tested this with the dev branch vs the 2.0 release and still receive the same error with just my single converter. Do you know of any way to add additional logging to the converter itself?

irakhlin, 2025-01-13:
@DJJo14 , my error is a bit different from yours. My converter is indeed in the /data/external_converters directory as you can see from the error:
```
z2m: Failed to call 'ExternalConverters' 'start' ...
```

the error in #317 seems to be Failed to call **'ExternalExtensions'**

ptvoinfo, 2025-01-24:
Not confirmed.

closed 2025-01-24 by ptvoinfo