ptvoinfo/zigbee-configurable-firmware #320: Personnal converter not good for new Z2M v2.0

alainvdu69, 2025-01-10:
Hi !
I've tried to add personnel converter generated by PTVO in the new version of Z2M : v2.0.0-2
I've a message when i try to save the code of the converter in the GUI of Z2M : 
 invalide code: module is not a constructor

If i remove the last line that contain 'module', i have another error.

ptvoinfo, 2025-01-11:
Please place the generated converter directly to the necessary folder without Z2M GUI.

alainvdu69, 2025-01-13:
i've try, with the 2025 version.
and restarted Z2M.
error :

[2025-01-13 16:12:35] error: 	z2m: Failed to call 'ExternalExtensions' 'start' (TypeError: module is not a constructor
    at ExternalExtensions.loadJS (/app/lib/extension/externalExtensions.ts:43:13)
    at processTicksAndRejections (node:internal/process/task_queues:105:5)
    at ExternalExtensions.loadFiles (/app/lib/extension/externalJS.ts:156:13)
    at ExternalExtensions.start (/app/lib/extension/externalJS.ts:45:9)
    at Controller.callExtensions (/app/lib/controller.ts:368:17)
    at Controller.start (/app/lib/controller.ts:186:9)
    at start (/app/index.js:161:5)
    at Timeout.restart [as _onTimeout] (/app/index.js:47:5))

[ptvo.volet.hall3.js.txt](https://github.com/user-attachments/files/18398570/ptvo.volet.hall3.js.txt)

alainvdu69, 2025-01-13:
i've only this converter installed : 
[image]

alainvdu69, 2025-01-13:
lol
i've discovered the mistake
external_extensions   vs   external_converters ...

closed 2025-01-13 by alainvdu69