ptvoinfo/zigbee-configurable-firmware #321: External converter with Z2M v2

alainvdu69, 2025-01-16:
Hi !

With external_converter (i've used 1st version  PTVO for 2025) , i’ve plenty of entity in HA :
– i’ve the binary_sensor with a huge difference : the default state should be off but it is on (it's reversing the reality) !
– i have select.xxxx_bind_command_xxxx entities with a “unknown” state
– i have select.xxxx_link_to_output_xxxx entities with a “unknown” state
– i have select.xxxx_switch_actions_xxxx entities with a “unknown” state
– i have select.xxxx_switch_type_xxxx entities with a “unknown” state

Before the v2 of Z2M and without external converter i've only the binary_sensor, and state were according to the reality.
After the installation of Z2M v2, the binary_sensor were unavailable/unknown and i've made MQTT template to recover.

My questions are on what to modify in the converter to :
– reverse the result of the binary_sensor
– remove the unwanted entities

Thanks.

alainvdu69, 2025-01-20:
it's strange : Z2M says "false" for l1, l2, l3, l4

[image]

and binary_sensor.xxxx are all to "on"

[image]

ptvoinfo, 2025-01-21:
These entities are used to change an input configuration. You can remove and ignore it. Also, you can remove this feature from the converter. Remove all occurrences like:
```
...ptvo_on_off_config_exposes('l1'),
```

alainvdu69, 2025-01-21:
Ok, it's work, thanks.

And about value inversion of the binary_sensor : 'false' in zigbee wich give 'on' in HA ?

ptvoinfo, 2025-01-24:
@alainvdu69 Sorry, but I do not know. It is an integration problem between Z2M and HA.

closed 2025-01-24 by alainvdu69