ptvoinfo/zigbee-configurable-firmware #322: ZHA  hint Feeding watchdog

wen-xiaoqiang, 2025-01-18:
I add a device，prompt Feeding watchdog in ZHA。

[image]

[image]

[image]

Please help me find where the problem is

ptvoinfo, 2025-01-18:
This is a message from HA. The watchdog timer in the firmware does not relate to this message.

Does the status light flash (your device works correctly)?

wen-xiaoqiang, 2025-01-18:
> 这是来自 HA 的消息。固件中的看门狗计时器与此消息无关。
> 
> 状态指示灯是否闪烁（您的设备正常工作）？

It's normal, it flashes once a second.

ptvoinfo, 2025-01-19:
Sorry, but I do not have other ideas. What module with CC2530 are you using?

ptvoinfo, 2025-01-19:
It seems, CC2530-PA2 contains a power amplifier (PA2), but you've configured the firmware for a simple CC2530.

closed 2025-02-09 by ptvoinfo