ptvoinfo/zigbee-configurable-firmware #324: MODBUS RTU not working since upgrade to Z2M 2.0.0.2

kds69, 2025-01-25:
Tried to use latest firmware and reprogramming. In HA, sensors are created but no value get through.
I can see that the CC22530+ CC2591 is pulling modbus data as it used to (via a MODBUS TTL). Don't have log showing error in Z2M add-on, 
(no external converter used).
Just LQI is getting through.

[image]

ptvoinfo, 2025-02-09:
Sorry about the delay with my answer. Do you the corresponding values in on the "State" tab in Z2M?

kds69, 2025-02-10:
> Sorry about the delay with my answer.

no pb.

> Do you the corresponding values in on the "State" tab in Z2M?

alas, values are not reported. I also checked directly on Mosquitto (MQTT test), and the values are not part of messages.

kds69, 2025-02-16:
did you have a chance to investigate ?

ptvoinfo, 2025-02-23:
@kds69 Please download the latest version from the website. I've fixed this issue.

kds69, 2025-03-15:
Thanks but still not good (checking both with zigbee2mqtt and MQTT-Explorer.
no value is reported in MQTT mosquitto.
I am sure the communication is happening between CC2530/CC2591 and HA/zigbee2mqtt as I can drive the refresh frequency (changing from every 10s to 5s for instance) and see the effect on UART/MODBUS TTL.
LEDs would blink as communication with MODBUS device goes, and this repeatedly every 10s or 5s according to refresh rate I impose.
Can I help in a way or another to solve this with you?