ptvoinfo/zigbee-configurable-firmware #330: PSM Wakeup

daestx, 2025-03-01:
Hi,
I have a question about the PSN Wakeup functionality. I activated the PSM mode and configured P10 as external wakeup,
On this input pin, I connected an external pullup resistor to VCC as well as a connection to a GPIO pin of my external Controller, which I'd like to use to wakeup my CC2530

here my configuration:

[image]

I'd expect that I can wakeup the Zigbee device as soon as I pull the pin to LOW. But this seems not to be the case. I observe that the chip is awake if only the external pull up is connected. Then I still receive data.

If I put the resistor as pull down, then it seems to works. No data transmission if P10 is not touched. As soon as P10 is put to high (just before data transmission) CC2530 wakes up and sends the expected data.

Please clarify. From the documentation, this is not clear to me

daestx, 2025-03-01:
Hmm, I'm not sure what I did wrong, but now it's working as expected

[image]

I pull the wakeup signal down for 500 ms. Then after a short period of 2ms I start the transmission. This is working. I'd expect, that with the falling edge of the wakeup signal the CC2530 starts to wake up. What are the constraints from timing point of view to safely wake up the chip

ptvoinfo, 2025-03-03:
@daestx 
I cannot suggest precise timings because I didn't measure it.

Obviously, the minimal interval is 4 us. In my projects, I use 1 ms delay.

UART uses interrupts, so I suppose, you can send UART data after 4us.