ptvoinfo/zigbee-configurable-firmware #331: new version 2.1.3-1 of Z2M break loading of external converter

alainvdu69, 2025-03-02:
Hi !

I've just installed the ZigBee2MQTT v2.1.3-1 and there is an error with external converter.
The log say : 
[2025-03-02 22:00:04] error: 	z2m: Failed to call 'ExternalConverters' 'start' (/app/data/extension/ptvo.volet.hall3.js:14
fz.legacy = ptvo_switch.meta.tuyaThermostatPreset;
                             ^
TypeError: Cannot read properties of undefined (reading 'tuyaThermostatPreset')
    at /app/data/extension/ptvo.volet.hall3.js:14:30
    at Script.runInContext (node:vm:149:12)
    at Script.runInNewContext (node:vm:154:17)
    at runInNewContext (node:vm:310:38)
    at ExternalConverters.loadModuleFromText (/app/lib/extension/externalJS.ts:187:24)
    at ExternalConverters.loadFiles (/app/lib/extension/externalJS.ts:156:52)
    at ExternalConverters.start (/app/lib/extension/externalJS.ts:45:20)
    at processTicksAndRejections (node:internal/process/task_queues:105:5)
    at Controller.callExtensions (/app/lib/controller.ts:368:17)
    at Controller.start (/app/lib/controller.ts:186:9))

And the UI of Z2M say "device not supported".

Reverting to Z2M v2.1.1-1 : it's fine again.

daestx, 2025-03-02:
Yeah this is a known topic. since version 2.0.0 converters need to be stored in a subfolder

[image]

please have a look here https://community.home-assistant.io/t/z2mqtt-external-converter/822123. The entry in the configuration.yaml can be removed

ptvoinfo, 2025-03-03:
Delete the following string in your converter.
```
fz.legacy = ptvo_switch.meta.tuyaThermostatPreset;
```

alainvdu69, 2025-03-03:
> Yeah this is a known topic. since version 2.0.0 converters need to be stored in a subfolder
> 
> [image]
> 
> please have a look here https://community.home-assistant.io/t/z2mqtt-external-converter/822123. The entry in the configuration.yaml can be removed
> 
> At least, i have no problems with the current update

this is not the problem ... i've the v2.0 since 2 months and the converter is in the right place

alainvdu69, 2025-03-03:
> Delete the following string in your converter.
> 
> ```
> fz.legacy = ptvo_switch.meta.tuyaThermostatPreset;
> ```

done ... and solved.
thanks a lot.

closed 2025-03-03 by alainvdu69