ptvoinfo/zigbee-configurable-firmware #332: Keypad (matrix keyboard)

kvs2160, 2025-03-10:
Hello.
I need a keyboard with 18 buttons (6x3).
After downloading the file, the battery capacity shows 100% and does not change when the battery is discharged. The status remains offline even after a day. I am sending links to photos.
What am I doing wrong?

[image]
[image]

ptvoinfo, 2025-03-20:
1. Please try to add the "Uptime" output sensor
2. Enable periodic reports.
3. Disable changing the report interval remotely.

kvs1960kvs, 2025-03-20:
I just checked my settings as you suggested. It doesn't work. What should I do?

kvs1960kvs, 2025-03-20:
I noticed that the firmware time and date are not updated.
[image]

kvs1960kvs, 2025-03-21:
If you prepare a hex program from May 23, 2024, then the battery % shows correctly. But if from 03/19/2025, then it does not work. The status is always offline! If you connect the computer battery after the load, then when the power decreases, the consequences will not change. Maybe the problem is in the program?

ptvoinfo, 2025-03-21:
Please try the updated released today. I've addressed to this issue.
https://ptvo.info/download/ptvo-firmware-2025-03-21.zip

kvs1960kvs, 2025-03-23:
Проверил. Сейчас все работает. Спасибо.

closed 2025-03-26 by ptvoinfo