ptvoinfo/zigbee-configurable-firmware #335: matrix keyboard

kkn1976, 2025-03-20:
Hello.
I have the same problem with the matrix keyboard as the question asked a few days ago (#332). The status is always offline and the battery level is always 100%!
Please explain it to me and point out any errors on this issue.
Thank you.

ptvoinfo, 2025-03-20:
1. Please try to add the "Uptime" output sensor
2. Enable periodic reports. 
3. Disable changing the report interval remotely.

kkn1976, 2025-03-20:
Here are my settings. With these settings, the status is always offline and the battery level is always 100%!
I noticed that if you make 12 buttons with these settings, then everything works. But I have 18-button remotes ready.

[image]
[image]

ptvoinfo, 2025-03-20:
Here is my keypad state with your settings but without real hardware buttons. The periodic report interval is 120 seconds (for faster battery state updates). 
[image]

kkn1976, 2025-03-20:
It doesn't work for me. Can you give me your hex for verification?

ptvoinfo, 2025-03-21:
@kkn1976 
Please try the updated released today. I've addressed to this issue.
https://ptvo.info/download/ptvo-firmware-2025-03-21.zip

kvs1960kvs, 2025-03-30:
[image]
Здравствуйте.
Я проверил версию “ptvo-firmware-2025-03-21”. 
1.	Батарея стала отслеживаться при снижении напряжения на ней.
2.	Доступность в сети появилась, но при отключении питания (без батарейки) даже на 48 часов, плата с клавиатурой остается в сети.
3.	Если выбирать ”время работы”, то из 18 кнопок работают кнопки single_l1 - single_l12, остальные 6 кнопок при нажатии просто определяются как single. Если не выбирать ”время работы”, то работают все 18 (single_l1 - single_l18).
4.	Заметил, что через несколько часов без нажатийй, кнопки не нажимаются, а батарея, сила сигнала и т.д. отслеживаются.
Проверял на 4 одинаковых платах. Все платы на 18 кнопок.

ptvoinfo, 2025-06-19:
Fixed in the latest version

closed 2025-06-19 by ptvoinfo