ptvoinfo/zigbee-configurable-firmware #343: How to configure a pairing button with RF-BM-2652P2 module

christopheVia, 2025-06-18:
Hello,
I'm working on a project using an RF-BM-2652P2 module intended to act as a router.
In the "General Information" section of the online documentation, you explain that the pairing button must be connected to input 1. Since pin 1 of the cc2652P is not user-accessible with the RF-BM-2652P2 module, how do I configure the firmware configurator exactly?
Someone asked this question in the documentation comments, but I didn't quite understand your answer.

ptvoinfo, 2025-06-19:
You should define the button on any pin in the "Input 1" row in the firmware configuration. The "Switch" mode for this button should be disabled. "Input 1" can also be configured as "External wake-up".

christopheVia, 2025-07-01:
Thank you for your reply. I still have a few questions to ask.
I was able to integrate the sensor into a Z2M network and connect it to the external converter created by the ptvo configurator. Here is a copy of the configuration created:

`Board type: CC2652P/CC1352P2
Device type: Router
Transmit enable (TXEN): P28, Enable high PA: P29
Manufacturer name: ptvo.info
Model ID: WaterZ
Status LED: P05, Joining or errors
Set default reporting interval (s): 60

Input pins:
P15: Input 1, GPIO, Pull-up (Bind command: On/Off)
P23: Input 2, ADC (accurate, max 1.15V)`

My questions:
1. I put a 330 ohm resistor + LED in series, between GND and P05 to serve as the status LED. The LED has fast blinks when the router is not yet paired (that's OK), but once paired, I only get a brief pulse of light at startup. This doesn't seem to fit the normal operation described in the documentation (Short, long blinks one per 4 seconds). Is my wiring and ptvo configuration corrects ?
2. The ADC reads the voltage supplied by a QDY30A analog pressure sensor. Would it be possible to convert this voltage to a water height or volume of water and use the external converter to set the new unit of measurement and the conversion constant, with a box to enter its value in Z2M for example ?

christopheVia, 2025-07-06:
I answer to myself. I found a solution to convert ADC raw data to volume and display this volume data in in the sensor expose page in Z2M. Here is the code :

```javascript
const zigbeeHerdsmanConverters = require('zigbee-herdsman-converters');
const zigbeeHerdsmanUtils = require('zigbee-herdsman-converters/lib/utils');

const exposes = zigbeeHerdsmanConverters['exposes'] || require("zigbee-herdsman-converters/lib/exposes");
const ea = exposes.access;
const e = exposes.presets;
const modernExposes = (e.hasOwnProperty('illuminance_lux'))? false: true;

const fz = zigbeeHerdsmanConverters.fromZigbeeConverters || zigbeeHerdsmanConverters.fromZigbee;
const tz = zigbeeHerdsmanConverters.toZigbeeConverters || zigbeeHerdsmanConverters.toZigbee;

const ptvo_switch = (zigbeeHerdsmanConverters.findByModel)?zigbeeHerdsmanConverters.findByModel('ptvo.switch'):zigbeeHerdsmanConverters.findByDevice({modelID: 'ptvo.switch'});
fz.ptvo_on_off = {
  cluster: 'genOnOff',
  type: ['attributeReport', 'readResponse'],
  convert: (model, msg, publish, options, meta) => {
      if (msg.data.hasOwnProperty('onOff')) {
          const channel = msg.endpoint.ID;
          const endpointName = `l${channel}`;
          const binaryEndpoint = model.meta && model.meta.binaryEndpoints && model.meta.binaryEndpoints[endpointName];
          const prefix = (binaryEndpoint) ? model.meta.binaryEndpoints[endpointName] : 'state';
          const property = `${prefix}_${endpointName}`;
	  if (binaryEndpoint) {
            return {[property]: msg.data['onOff'] === 1};
          }
          return {[property]: msg.data['onOff'] === 1 ? 'ON' : 'OFF'};
      }
  },
};

const switchTypesList = {
    'switch': 0x00,
    'single click': 0x01,
    'multi-click': 0x02,
    'reset to defaults': 0xff,
};

const switchActionsList = {
    on: 0x00,
    off: 0x01,
    toggle: 0x02,
};

const inputLinkList = {
    no: 0x00,
    yes: 0x01,
};

const bindCommandList = {
    'on/off': 0x00,
    'toggle': 0x01,
    'change level up': 0x02,
    'change level down': 0x03,
    'change level up with off': 0x04,
    'change level down with off': 0x05,
    'recall scene 0': 0x06,
    'recall scene 1': 0x07,
    'recall scene 2': 0x08,
    'recall scene 3': 0x09,
    'recall scene 4': 0x0A,
    'recall scene 5': 0x0B,
    'dimmer': 0x0C,
    'dimmer (hue)': 0x0D,
    'dimmer (saturation)': 0x0E,
    'dimmer (color temperature)': 0x0F,
    'intruder alarm systems (ias)': 0x20,
};
[... 126 lines cut ...]
    ...ptvo_on_off_config_exposes('l1'),
    exposes.numeric('l2_raw', ea.STATE).withDescription('Raw ADC data'),
    exposes.numeric('l2_volume', ea.STATE).withDescription('calculated volume').withUnit('l'),
    exposes.numeric('Conversion_coef', ea.STATE).withDescription('Conversion coefficient, from ADC to volume (liters)'),
  ],
  meta: {
    multiEndpoint: true,
  },
  endpoint: (device) => {
    return {
      l1: 1,
      l2: 2,
    };
  },
    configure: async (device, coordinatorEndpoint, logger) => {
      const endpoint = device.getEndpoint(1);
      await endpoint.read('genBasic', ['modelId', 'swBuildId', 'powerSource']);
    },

};

module.exports = device;
``` 

So far, everything seems to be working normally. However, it is necessary to modify this JS file to correct the conversion coefficient. Is there a way to save this value inside the sensor programmed with PTVO in order to save it permanently, and possibly be able to modify it from the Z2M settings (Specifics) page ?