ptvoinfo/zigbee-configurable-firmware #349: Pulse switch impulse duration issue

ncolomer, 2025-08-15:
I'm using [Pulse switch](https://ptvo.info/zigbee-configurable-firmware-features/pulse-switch/) on a PTVO firmware (2024-07-23) flashed on a CC2530 to activate relays for a period of time.

<details><summary>Firmware configuration</summary>
<p>

```
Board type: CC2530
Device type: Router
Model ID: garage.door
Update firmware's timestamp : 2024-09-26
Disable periodic On/Off reports for GPIO outputs: Yes
Status LED: P01, Sending reports
Set default reporting interval (s): 0

Output pins:
P02: Output 2, Pulse switch, Pull-up (Impulse: 500, : ms, Restore after power off: Off, Change to state after interval: Off, Interval: 500, : ms, Linked input: Restart impulse timer), Remember state
P04: Output 3, Pulse switch, Pull-up (Impulse: 500, : ms, Restore after power off: Off, Change to state after interval: Off, Interval: 500, : ms, Linked input: Restart impulse timer), Remember state
P05: Output 4, Pulse switch, Pull-up (Impulse: 500, : ms, Restore after power off: Off, Change to state after interval: Off, Interval: 500, : ms, Linked input: Restart impulse timer), Remember state
P06: Output 5, Pulse switch, Pull-up (Impulse: 500, : ms, Restore after power off: Off, Change to state after interval: Off, Interval: 500, : ms, Linked input: Restart impulse timer), Remember state

Input pins:
P00: Input 1, External wake-up, Pull-up
```

</p>
</details> 

My various tests showed anytime I set an `Impulse` value to more than 1s (eg. 5s) the effective impulse is something less than 1s.

It might be related to internal timers capabilities and/or usage by the firmware.

Can you confirm?

Side note: it would be great if we could tune this value OTA using Zigbee command. It would avoid reflashing the firmware.

ptvoinfo, 2025-08-20:
Obviously, I cannot discuss any issues of the old version. It is possible they were already fixed in the latest version.

ptvoinfo, 2025-08-21:
@ncolomer Could you try the latest version released today?